openhwfoundation/cva6 #717: Steps to completely disable the data cache

E-Poulin, 2021-09-01:
As agreed in the meeting, these are the steps that you need to do, to be able to completely disable the data cache.

Firstly : You need a parameter to generate caches in the ariane_pck.sv.
https://github.com/E-Poulin/cva6-softcore-contest/blob/d491e67ae04ad4e492a3ae6f5af401e8d77955b4/include/ariane_pkg.sv#L428

Secondly : You need conditions (if generate) to generate or not the caches, in the wt_cache_subsystem (for the WT cache), and in the std_cache_subsystem (for the WB cache).
https://github.com/E-Poulin/cva6-softcore-contest/blob/d491e67ae04ad4e492a3ae6f5af401e8d77955b4/src/cache_subsystem/wt_cache_subsystem.sv#L116-L165

Thirdly : You need to direct the core requests from the LSU to the Axi bus (and then to the main memory).  I did not finish this part because of lack of time.
https://github.com/E-Poulin/cva6-softcore-contest/blob/d491e67ae04ad4e492a3ae6f5af401e8d77955b4/src/cache_subsystem/wt_cache_subsystem.sv#L194-L215

Regards,
Emeric

MikeOpenHWGroup, 2023-02-17:
Hi @E-Poulin, this Issue is very old, do you know its status?

@ASintzoff, this looks like information that should be in a Design Specification.

E-Poulin, 2023-02-17:
> Hi @E-Poulin, this Issue is very old, do you know its status?
> 
> @ASintzoff, this looks like information that should be in a Design Specification.

Hello Mike,

I'm sorry but since I left Thales in September 2021, I haven't worked on the CVA6, so I don't know if someone worked on this issue since all this time.

However, with a quick look at the current sources and packages files of the CVA6, I didn't see any way to completely disable the data cache.

Best regards,
Emeric

JeanRochCoulon, 2023-03-14:
Hello,
We should want to disable the Dcache in the coming months. Of course the steps described by @E-Poulin will be followed. The step 3 was the more interesting step. As it is not covered, it not worth keeping the issue open, I propose to close it.

closed 2023-03-14 by JeanRochCoulon