openhwfoundation/cva6 #721: Failure to detect cache coherency violation.

zarubaf, 2021-09-03:
Location: cva6/src/cache_subsystem (https://github.com/openhwgroup/cva6/tree/master/src/cache_subsystem).

Details: According to the RISC-V specification, when the instruction memory is modified, the software should handle the cache coherency using FENCE.I instruction. If the input program modifies the instruction memory but does not use a FENCE.I instruction, there is a mismatch in the trace log of Ariane and spike. Ariane executes the instructions from the old instruction memory whereas spike executes the instruction from the newly updated instruction memory.

jquevremont, 2021-09-06:
The input and log files that were sent by the team who discovered the issue.
[bug_3.zip](https://github.com/openhwgroup/cva6/files/7114423/bug_3.zip)

Phantom1003, 2022-05-13:
Hi, we believe this is not a bug. The reason causes this problem is that cva6 has cache while spike doesn't. If the software modifies the code segment without a fence.i, cva6 will continue using the old instructions in the cache.

MikeOpenHWGroup, 2023-02-17:
Thanks for your insights @Phantom1003.

@zarubaf, please have a look at this an confirm whether or not this issue is a real bug.  Thanks!

zarubaf, 2023-10-20:
> Hi, we believe this is not a bug. The reason causes this problem is that cva6 has cache while spike doesn't. If the software modifies the code segment without a fence.i, cva6 will continue using the old instructions in the cache.

@Phantom1003 is 100% correct here. The testprogram is not valid, if you want to see updated values on `0x800000e0` you need to execute a `fence.i`. Instruction and data streams are not coherent in RISC-V.

closed 2023-10-20 by zarubaf