openhwfoundation/cva6 #722: How to run a program without operating system on FPGA

2536786901, 2021-09-05:
I use the genesys2 development board, how to run my own program without operating system

Bill94l, 2021-09-06:
To be able to run and debug a software application in standalone on the CVA6 core, we used the RISC-V GNU Compiler for coss-compilation, once the binary is generated you can use GNU Debugger (GDB) and Open On-Chip Debugger (OpenOCD) for binary loading and execution control.

2536786901, 2021-09-07:
Thank you for your answer. I just learned this and I am using the genesys2 development board. I would like to ask if the operating system is not loaded, what are the specific steps to use openOCD to directly operate the program?

Bill94l, 2021-09-08:
Hello!

- Did you follow all the steps mentioned in the README ? (especially the FPGA Emulation step)
- Linux should work normally, did you change something in the hardware architecture?
- To use openOCD and gdb, follow the debugging steps available on the README, and here are the steps that i followed:

## OpenOCD
Sur Ubuntu, assurez-vous que tout est installé avec :
```
$ sudo apt install make libtool pkg-config autoconf automake texinfo
```
De plus, vous devez configurer les bibliothèques libusb et libftdi. Sur Ubuntu :
```
$ sudo apt install libusb-1.0-0-dev libftdi1-dev
```
Une fois toutes les dépendances installées, OpenOCD peut être configuré.

- Sources de téléchargement :
```
$ git clone https://github.com/riscv/riscv-openocd
$ cd riscv-openocd
```
- Préparez un répertoire de compilation :
```
$ mkdir build
```
- Lancez le script d'amorçage :
```
$ ./bootstrap
```
- Lancer la configuration :
```
$ ./configure --enable-ftdi --prefix=<absolute path>/build --exec-prefix=<absolute path>/build
```
- Compilez et installez les fichiers :
```
$ make
$ make install
```
Une fois l'installation terminée, n'oubliez pas d'ajouter riscv-openocd/build/bin à votre PATH.
```
$ export PATH=$PATH:<path to riscv-openocd>/build/bin
```
## câble HS2
Il est nécessaire d'ajouter une règle udev pour utiliser le câble. OpenOCD fournit un fichier contenant la règle dont nous avons besoin. Copiez-le dans /etc/udev/rules.d/
```
$ sudo cp <path to riscv-openocd>/build/share/openocd/contrib/60-openocd.rules /etc/udev/rules.d
```
Le fichier est également disponible ici : https://github.com/riscv/riscv-openocd/blob/riscv/contrib/60-openocd.rules L'entrée particulière concernant le câble HS2 est :
```
ATTRS{idVendor}=="0403", ATTRS{idProduct}=="6014", MODE="660", GROUP="plugdev", TAG+="uaccess"
```
Ensuite, redémarrez votre système ou rechargez la configuration udev avec :
```
$ sudo udevadm control --reload
```
Pour vérifier si le câble est reconnu, exécutez lsusb. Il devrait y avoir une ligne comme celle-ci :
```
$ lsusb
```
```
Bus 005 Device 003: ID 0403:6014 Future Technology Devices International, Ltd FT232HSingle HS USB-UART/FIFO IC
```

Cordially
Billal IGHILAHRIZ

2536786901, 2021-09-08:
First of all, thank you very much for your patience. I am sorry that my question is not clear. Now I have compiled my own program using the riscv tool chain. My question is how to make the compiled program start and run in the SD card of genesys2 without being based on any operating system .

 

  
 
 
  
 
  
  $ ./configure --enable-ftdi --prefix=<absolute path&gt;/build --exec-prefix=<absolute path&gt;/build   
 $ export PATH=$PATH:<path to riscv-openocd&gt;/build/bin  
 
 $ sudo cp <path to riscv-openocd&gt;/build/share/openocd/contrib/60-openocd.rules /etc/udev/rules.d

Bill94l, 2021-09-08:
-  You can use OpenOCD and GDB  to make the compiled program start and run 
- The binary is loaded from JTAG(not from SD card), this requires[ JTAG-HS2](https://store.digilentinc.com/jtag-hs2-programming-cable/) programming cable

## Launch steps
- Dans un terminal, lancez OpenOCD :
```
$ openocd -f fpga/openocd_digilent_hs2.cfg
```

- S'il réussit, vous devriez voir :
```
Open On-Chip Debugger 0.10.0+dev-00832-gaec5cca (2019-12-10-14:21)
Licensed under GNU GPL v2
For bug reports, read
    http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x249511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x2)
Info : datacount=2 progbufsize=8
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=64, misa=0x40141105
Info : Listening on port 3333 for gdb connections
Ready for Remote Connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```
- Dans un terminal séparé, lancez gdb :
```
$ riscv64-unknown-elf-gdb helloworld.riscv
```
- vous devez utiliser le gdb de la chaîne d'outils RISC-V. S'il réussit, vous devriez voir :
```
GNU gdb (GDB) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from helloworld.riscv...
(gdb) 
```
- Dans gdb , vous devez connecter gdb à openocd :
```
(gdb) target remote :3333
```
- s'il réussit, vous devriez voir la connexion gdb dans openocd :
```
Info : accepting 'gdb' connection on tcp/3333
```
- Dans gdb , chargez helloword.riscv sur la plate-forme FPGA par la commande load  :
```
(gdb) load
Loading section .vectors, size 0x80 lma 0x80000000
Loading section .init, size 0x60 lma 0x80000080
Loading section .text, size 0x19010 lma 0x800000e0
Loading section .rodata, size 0x1520 lma 0x800190f0
Loading section .eh_frame, size 0x50 lma 0x8001a610
Loading section .init_array, size 0x4 lma 0x8001a660
Loading section .data, size 0x9d4 lma 0x8001a668
Loading section .sdata, size 0x40 lma 0x8001b040
Start address 0x80000080, load size 110712
Transfer rate: 63 KB/sec, 7908 bytes/write.
```
- Enfin, dans gdb , vous pouvez lancer l'application coremark par commande c:
```
(gdb) c
Continuing.
(gdb) 
```
I think you have all the tools that you need to start your first program.

Good luck
BI

2536786901, 2021-09-09:
I would like to express my heartfelt thanks to you again, I follow your process:
(gdb) c
Continuing.
Then there is no response. I compiled riscv-unknown-linux-gnu-g++, and then used riscv-unknown-linux-gnu-gdb. Should I use the riscv64-unknown-elf-gcc version?

  

 
  
  
  Open On-Chip Debugger 0.10.0+dev-00832-gaec5cca (2019-12-10-14:21) Licensed under GNU GPL v2 For bug reports, read     http://openocd.org/doc/doxygen/bugs.html Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport&gt;'. Info : clock speed 1000 kHz Info : JTAG tap: riscv.cpu tap/device found: 0x249511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x2) Info : datacount=2 progbufsize=8 Info : Examined RISC-V core; found 1 harts Info :  hart 0: XLEN=64, misa=0x40141105 Info : Listening on port 3333 for gdb connections Ready for Remote Connections Info : Listening on port 6666 for tcl connections Info : Listening on port 4444 for telnet connections

Bill94l, 2021-09-09:
- Yes, you should use the `riscv64-unknown-elf-gcc`
- From #444  you would need to develop a BSP.
## BSP (Board Support Package)
The **BSP** is a low-level software package that supports the hardware platform.  it consists of three directories:
- **Config**: It contains the linker script and file which defines some constant relating to the FPGA platform.
- **Drivers**: it contains all peripheral driver that we use. 
- **Hal**: Hardware Abstract Layer contains the runtime system interrupt vectors, system calls.

For the driver of the bsp you can refer to **fpga/bootrom**

2536786901, 2021-09-09:
Thank you very much for your answers

I use riscv64-unknown-elf-g++ to compile helloworld.cpp, use "c" to display continuing, but no results are displayed, and use "r" to display The "remote" target does not support "run". How can I test The performance of the algorithm?

(gdb) r
The "remote" target does not support "run".  Try "help target" or "continue".
(gdb) continue
Continuing.

Moschn, 2021-09-09:
As far as I understand you want to run something bare-metal on the FPGA. First of all, you have to tell your compiler/linker to not use any underlying library such as Glibc: https://github.com/openhwgroup/cva6/blob/f5cbaa92e9b21dbd52d0de7851069e979d2f011d/fpga/src/bootrom/Makefile#L13
Secondly, no print statement will be available to you as soon as you activate these options (i.e., the compilation should fail) since you do not have an underlying OS that provides a driver for a print. Our bootrom, for example, contains a driver for the UART, SPI, and SD card. Thirdly, you also need to provide a linker file (see https://github.com/openhwgroup/cva6/blob/master/fpga/src/bootrom/linker.lds) to specify at which address your executable should be loaded to and what the start point is. For the start point, you actually have to write some assembly that sets up some stack and then jumps to the main routine you want to call. (see https://github.com/openhwgroup/cva6/blob/master/fpga/src/bootrom/startup.S).

The entire process is A LOT easier in simulation because we can use the proxy kernel in that case.

Now to your specific questions (which I think are actually not the problem why your HelloWorld does not run). To load a program onto the FPGA using GDB you have to do the following steps (copied from @Bill94l above):
```console
$ openocd -f fpga/openocd_digilent_hs2.cfg
Open On-Chip Debugger 0.10.0+dev-00832-gaec5cca (2019-12-10-14:21)
Licensed under GNU GPL v2
For bug reports, read
    http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : clock speed 1000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x249511c3 (mfg: 0x0e1 (Wintec Industries), part: 0x4951, ver: 0x2)
Info : datacount=2 progbufsize=8
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=64, misa=0x40141105
Info : Listening on port 3333 for gdb connections
Ready for Remote Connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections

$ riscv64-unknown-elf-gdb
GNU gdb (GDB) 9.1
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-pc-linux-gnu --target=riscv64-unknown-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...

(gdb) target remote :3333

(gdb) load helloworld.elf
Loading section .vectors, size 0x80 lma 0x80000000
Loading section .init, size 0x60 lma 0x80000080
Loading section .text, size 0x19010 lma 0x800000e0
Loading section .rodata, size 0x1520 lma 0x800190f0
Loading section .eh_frame, size 0x50 lma 0x8001a610
Loading section .init_array, size 0x4 lma 0x8001a660
Loading section .data, size 0x9d4 lma 0x8001a668
Loading section .sdata, size 0x40 lma 0x8001b040
Start address 0x80000080, load size 110712
Transfer rate: 63 KB/sec, 7908 bytes/write.

(gdb) continue
```

The `load` command should load the elf file to the correct address and set the program counter to `_start`. `c` or `continue` then just allows the core to start executing. `run` is not supported in an attached GDB session! Also the terminal with the GDB session will most likely never output anything from your program. If you want to print something to UART you can use something like 
```console
$ screen /dev/ttyUSB0 115200
```

MikeOpenHWGroup, 2023-02-17:
Hi @2536786901, the last comment on this Issue was more than 1.5 years ago and it was a "thumbs-up" so I will assume the issue is resolved and will close it.   Thanks for your interest in CVA6!

closed 2023-02-17 by MikeOpenHWGroup