openhwfoundation/cva6 #730: problem on  work-ver/Variane_testharness rv64um-v-divuw

vulkomilev, 2021-09-17:
After the step 

 make verilate DEBUG=1

on step 

 work-ver/Variane_testharness rv64um-v-divuw

i get the following error

This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 41909
Variane_testharness: ../fesvr/elfloader.cc:21: std::map<std::__cxx11::basic_string<char>, long unsigned int> load_elf(const char*, memif_t*, reg_t*): Assertion `fd != -1' failed.

I am on ubuntu 20.04

hasanalshaikh, 2021-09-21:
I am getting the same error after I managed to make the binary after a lot of tinkering around. The binary refused to be made for any verilator version>4.20. The versions I have are riscv64-unknown-elf-gcc=>11.1.0 and verilator v4.014 . I have tried other older versions of verilator (e.g., 4.002, 4.004) but every time I got the same error as @vulkomilev

FranDuqueAyachi, 2022-05-01:
Hi guys!
I have the same error, Have you managed to fix the error?
thank you!

hhhsiang, 2023-02-14:
Hi, @vulkomilev @hasanalshaikh @FranDuqueAyachi , I also have the same error. Have you fix it?

MikeOpenHWGroup, 2023-02-17:
Hello @hhhsiang, @FranDuqueAyachi, @hasanalshaikh and @vulkomilev.   What is your current status?

By following the instructions in the top-level [README](https://github.com/openhwgroup/cva6#install-verilator-simulation-flow) I am able to compile without issues.  I am using Verilator 4.100, and `$RISCV/riscv64-unknown-linux-gnu-gcc -v` yields the following:
```
Using built-in specs.
COLLECT_GCC=/opt/riscv/bin//riscv64-unknown-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv64-unknown-linux-gnu/10.2.0/lto-wrapper
Target: riscv64-unknown-linux-gnu
Configured with: /home/mike/GitHubRepos/riscv/riscv-gnu-toolchain/main/riscv-gcc/configure --target=riscv64-unknown-linux-gnu --prefix=/opt/riscv --with-sysroot=/opt/riscv/sysroot --with-system-zlib --enable-shared --enable-tls --enable-languages=c,c++,fortran --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libsanitizer --disable-nls --disable-bootstrap --src=.././riscv-gcc --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-O2   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-O2   -mcmodel=medlow'
Thread model: posix
Supported LTO compression algorithms: zlib
gcc version 10.2.0 (GCC) 
```
I am running this on Ubuntu 22.04.

hhhsiang, 2023-02-21:
Hi @MikeOpenHWGroup 
I tried the command under cva6：$RISCV/riscv64-unknown-linux-gnu-gcc -v
it showed：-bash: /riscv64-unknown-linux-gnu-gcc: No such file or directory

The followings are the steps I build the environment：
At first, I used these command to checkout the repository under /home/aislab42190/：

git clone https://github.com/openhwgroup/cva6.git
cd cva6
git submodule update --init --recursive
after that, I build a new directory named "RISCV" under /home/aislab42190/
then I set the environment variable by these commands：
6. export RISCV="/home/aislab42190/RISCV/"
7. export PATH=$PATH:$RISCV/bin
8. source ~/.bashrc

to install all required tools：
9. ./ci/setup.sh
10. make verilate
11. make verilate DEBUG=1

after the above commands were executed, I use the command：
work-ver/Variane_testharness rv64um-v-divuw

then it showed：
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 36327
Variane_testharness: ../fesvr/elfloader.cc:23: std::map<std::__cxx11::basic_string, long unsigned int> load_elf(const char*, memif_t*, reg_t*): Assertion `fd != -1' failed.

Did I have wrong during setting the environment variable path?
I am running this on Ubuntu 20.04.

Thank you very much！

zarubaf, 2023-02-25:
That means the path to your binary `rv64um-v-divuw` isn't good (wrong path, binary doesn't exist). It can't open the file descriptor.

MikeOpenHWGroup, 2023-03-06:
Hi @hhhsiang, was @zarubaf's comment helpful?  Are you still having trouble?

MikeOpenHWGroup, 2023-03-09:
Thanks @hhhsiang, I will close this issue.