openhwfoundation/cva6 #732: Issue with Spike simulation for CSR test program

RanjanThales, 2021-09-24:
Hi,

I am trying run csr test program generated from **gen_csr_test.py**
(https://github.com/google/riscv-dv/blob/master/scripts/gen_csr_test.py
input yml files from : https://github.com/openhwgroup/core-v-verif/tree/master/lib/corev-dv)

I am seeing spike simulation going into exception trap_illigal_instruction (as follow)
[image]

Is there any known solution to avoid such errors !!

With Regards
Ranjan

MikeOpenHWGroup, 2021-09-24:
Hi @RanjanThales, can you confirm that the access causing the trap is a `csrrw` of CSR `instret`?

The privileged spec provides some rules for `instret` access.  Section 4.1.7 states:
```
When the CY, TM, IR, or HPMn bit in the scounteren register is clear, attempts to read the cycle,
time, instret, or hpmcountern register while executing in U-mode will cause an illegal instruction
exception. When one of these bits is set, access to the corresponding register is permitted.
```

So, I think that in order for the `csrrw tb, instret, a5` instruction to execute either the core must be in Supervisor mode _or_ the IR bit in `scounteren` must be set.

Note that the CSR access test script (gen_csr_test.py) has no knowledge of these rules and so these must be manually added to the test-program.

RanjanThales, 2021-09-29:
Thanks Mike.
I worked with the mentioned reference.
With some manual setting to **scounteren csr** from test-program, I could resolve **trap_illigal_instruction** and the Spike simulation started moving to next stage.

closed 2021-09-29 by RanjanThales