openhwfoundation/cva6 #736: error during make verilate

nFal, 2021-09-26:
I'm getting the following error while doing make verilate on a fresh checkout of the repo:

```
../corev_apu/tb/ariane_tb.cpp: In function ‘int main(int, char**)’:
../corev_apu/tb/ariane_tb.cpp:323:49: error: ‘class Variane_testharness’ has no member named ‘ariane_testharness__DOT__i_sram__DOT__gen_cut__BRA__0__KET____DOT__gen_mem__DOT__i_ram__DOT__Mem_DP’
  323 |   memif.read(0x80000000, mem_size, (void *)top->ariane_testharness__DOT__i_sram__DOT__gen_cut__BRA__0__KET____DOT__gen_mem__DOT__i_ram__DOT__Mem_DP);
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Variane_testharness.mk:68: recipe for target 'ariane_tb.o' failed
make[1]: *** [ariane_tb.o] Error 1
make[1]: *** Waiting for unfinished jobs....
```

My verilator version:
`Verilator 4.213 devel rev v4.212-23-gd384a698`

zarubaf, 2021-09-26:
Verilator changed the way it names dot references and there is a pretty crude preloading model. If you switch to Verilator 4.002 (https://github.com/openhwgroup/cva6#tool-requirements) that should fix it.

Nevertheless, that needs fixing so thanks for raising the issue and I’ll keep it open.

zeeshanrafique23, 2021-11-07:
Hi @zarubaf, I am using the verilator version 4.002 but still the simulation got stuck while running the `cd work-ver && make -j -f Variane_testharness.mk` step in `make verilate` command.

zeeshanrafique23, 2021-11-07:
Also the fusesoc simulation is also not working in my case, may there is some change is directory structure or something...
@olofk

FranDuqueAyachi, 2022-04-30:
Hi!, I have exactly the same problem, 
have you been able to fix the error?

zeeshanrafique23, 2022-04-30:
Hi Fran,
Use verilator-4.110 version.
Check ci/setup.sh line 14.

FranDuqueAyachi, 2022-05-01:
> Hi Fran, Use verilator-4.110 version. Check ci/setup.sh line 14.

Hi! Thanks for your answer.
The problem is that I don't have sudo permissions on this machine, I had to install verilator from the file ./ci/install-verilator.sh, and the version I have is 4.221. Is there a way to install or downgrade without using sudo apt?
Thanks again!

zeeshanrafique23, 2022-05-01:
You can build verilator from source for the specific version and then include the bin folder to the PATH.

https://github.com/verilator/verilator

FranDuqueAyachi, 2022-05-01:
Thank you very much!, I have corrected the error and now I have the file Variane_testharness.
however when executing the command (work-ver/Variane_testharness rv64um-v-divuw
) I get the following error and the vcd file is not generated:

This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 46461
Variane_testharness: ../fesvr/elfloader.cc:23: std::map<std::__cxx11::basic_string<char>, long unsigned int> load_elf(const char*, memif_t*, reg_t*): Assertion `fd != -1' failed.

Again thanks for your help!

My version of verilator is now 4.110
Ubuntu 18.04

zeeshanrafique23, 2022-05-02:
You need to pass `DEBUG=1` during make to enable the trace or to see the waveforms.
https://github.com/openhwgroup/cva6/blob/b40bb3264bc0ca0b5b9e9a3eb351cbaaa9b50b62/Makefile#L608
https://github.com/openhwgroup/cva6#build-model-and-run-simulations

zeeshanrafique23, 2022-05-02:
Check this:
```
Example:
- run a bare metal test to generate a VCD waveform:
    ./Variane_testharness -v rv64ui-p-add.vcd $RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-add
```
here `rv64ui-p-add` is an elf file and `rv64ui-p-add.vcd` is output vcd file.

FranDuqueAyachi, 2022-05-03:
> 
Hi!
I actually made de make verilate DEBUG=1 without any error and I got the folder work-ver with all fields correctly installed.
However, when I try: work-ver/Variane_testharness rv64um-v-divuw  or even ur example: ./Variane_testharness -v rv64ui-p-add.vcd $RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-add
I got the same issue:
fduque@vlsiws16% ./Variane_testharness -v rv64ui-p-add.vcd $RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-add
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 43445
Variane_testharness: ../fesvr/elfloader.cc:22: std::map<std::__cxx11::basic_string<char>, long unsigned int> load_elf(const char*, memif_t*, reg_t*): Assertion `fd != -1' failed.
Thank you so much for your atention!

FranDuqueAyachi, 2022-05-03:
> Check this:
> 
> ```
> Example:
> - run a bare metal test to generate a VCD waveform:
>     ./Variane_testharness -v rv64ui-p-add.vcd $RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-add
> ```
> 
> here `rv64ui-p-add` is an elf file and `rv64ui-p-add.vcd` is output vcd file.

> Check this:
> 
> ```
> Example:
> - run a bare metal test to generate a VCD waveform:
>     ./Variane_testharness -v rv64ui-p-add.vcd $RISCV/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-p-add
> ```
> 
> here `rv64ui-p-add` is an elf file and `rv64ui-p-add.vcd` is output vcd file.

Thank you very much! it worked perfectly!
now, Im gonna try to keep following the next steps with my own hello world program using the pk and lets see if I can check it on Questasim!

hhhsiang, 2023-02-14:
Hi, @FranDuqueAyachi 
I have the same error after executing the command (work-ver/Variane_testharness rv64um-v-divuw)：
Variane_testharness: ../fesvr/elfloader.cc:23: std::map<std::__cxx11::basic_string, long unsigned int> load_elf(const char*, memif_t*, reg_t*): Assertion `fd != -1' failed.

How did you fix it?
Besides, I found that there is no "share" folder under $RISCV/riscv64-unknown-elf/
Did you have the same problem at that time?
Thanks for helping!

MikeOpenHWGroup, 2023-02-17:
Hi @hhhsiang

> Besides, I found that there is no "share" folder under $RISCV/riscv64-unknown-elf/

What was the command you used to generate `work-ver/Variane_testharness rv64um-v-divuw`?  It looks to me like you've got an environment variable pointing to the wrong place.

hhhsiang, 2023-02-17:
Hi @MikeOpenHWGroup 
At first, I used these command to checkout the repository under /home/aislab42190/：
1. git clone https://github.com/openhwgroup/cva6.git
2. cd cva6
3. git submodule update --init --recursive

after that, I build a new directory named "RISCV" under /home/aislab42190/.
then I set the environment variable by these commands：

4. export RISCV="/home/aislab42190/RISCV/"
5. export PATH=$PATH:$RISCV/bin
6. source ~/.bashrc

to install all required tools：
8. ./ci/setup.sh
9. make verilate
10. make verilate DEBUG=1

after the above commands were executed, I use the command：
work-ver/Variane_testharness rv64um-v-divuw

then it showed：
aislab42190@aislab42190:~/cva6$ work-ver/Variane_testharness rv64um-v-divuw
This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
Listening on port 36327
Variane_testharness: ../fesvr/elfloader.cc:23: std::map<std::__cxx11::basic_string<char>, long unsigned int> load_elf(const char*, memif_t*, reg_t*): Assertion `fd != -1' failed.

Did I have wrong during setting the environment variable path?

Thank you very much！

MikeOpenHWGroup, 2023-03-06:
Hmmm.  This looks like a duplicate of #730.  I will close this one and we can track progress on that issue.

closed 2023-03-06 by MikeOpenHWGroup