openhwfoundation/cva6 #738: Synthesis is failing

Saqlinahamad, 2021-09-30:
Hi
I had raised a issue to lattice semiconductor regarding synthesis,
They said :
Are the only files in the script make the synthesis of cva6 core, As when they are are checking each .sv file, they can't able to locate some defines that are written over the RTL Code.
So I am sending an attachment of script file, could you let me know these are only files required for synthesis.

Saqlinahamad, 2021-09-30:
prj_open cva6_riscv.rdf 
prj_set_opt VerilogStandard {"System Verilog"}
prj_set_impl_opt synthesis {lse}

prj_set_impl_opt {include path} ../../../git_subtrees/cva6/include
prj_set_impl_opt {include path} ../../../git_subtrees/cva6/src/util
prj_set_impl_opt {include path} ../../../git_subtrees/cva6/src/common_cells 
prj_set_impl_opt {include path} ../../../git_subtrees/cva6/src/axi/src/axi

prj_add_source ../../../git_subtrees/cva6/include/riscv_pkg.sv \
               ../../../git_subtrees/cva6/src/riscv-dbg/src/dm_pkg.sv \
               ../../../git_subtrees/cva6/include/ariane_pkg.sv \
               ../../../git_subtrees/cva6/include/std_cache_pkg.sv \
               ../../../git_subtrees/cva6/src/axi/src/axi_pkg.sv \
               ../../../git_subtrees/cva6/src/register_interface/register_interface/src/reg_intf.sv \
               ../../../git_subtrees/cva6/src/register_interface/register_interface/src/reg_intf_pkg.sv \
               ../../../git_subtrees/cva6/include/axi_intf.sv \
               ../../../git_subtrees/cva6/include/ariane_axi_pkg.sv \
               ../../../git_subtrees/cva6/include/wt_cache_pkg.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpnew_pkg.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpu_div_sqrt_mvp/hdl/defs_div_sqrt_mvp.sv \
               ../../../git_subtrees/cva6/src/util/instr_tracer_if.sv \
               ../../../git_subtrees/cva6/src/decoder.sv \
               ../../../git_subtrees/cva6/src/compressed_decoder.sv \
               ../../../git_subtrees/cva6/src/alu.sv \
               ../../../git_subtrees/cva6/src/ariane.sv \
               ../../../git_subtrees/cva6/src/serdiv.sv \
               ../../../git_subtrees/cva6/src/amo_buffer.sv \
               ../../../git_subtrees/cva6/src/branch_unit.sv \
               ../../../git_subtrees/cva6/src/cache_subsystem/cache_ctrl.sv \
               ../../../git_subtrees/cva6/src/commit_stage.sv \
               ../../../git_subtrees/cva6/src/controller.sv \
               ../../../git_subtrees/cva6/src/csr_buffer.sv \
               ../../../git_subtrees/cva6/src/csr_regfile.sv \
               ../../../git_subtrees/cva6/src/ex_stage.sv \
               ../../../git_subtrees/cva6/src/cache_subsystem/cva6_icache.sv \
               ../../../git_subtrees/cva6/src/id_stage.sv \
               ../../../git_subtrees/cva6/src/instr_realign.sv \
               ../../../git_subtrees/cva6/src/issue_read_operands.sv \
               ../../../git_subtrees/cva6/src/issue_stage.sv \
               ../../../git_subtrees/cva6/src/load_unit.sv \
               ../../../git_subtrees/cva6/src/load_store_unit.sv \
               ../../../git_subtrees/cva6/src/cache_subsystem/miss_handler.sv \
               ../../../git_subtrees/cva6/src/multiplier.sv \
               ../../../git_subtrees/cva6/src/axi_shim.sv \
               ../../../git_subtrees/cva6/src/ex_stage.sv \
               ../../../git_subtrees/cva6/src/mmu.sv \
               ../../../git_subtrees/cva6/src/mult.sv \
               ../../../git_subtrees/cva6/src/pmp/src/pmp_entry.sv \
               ../../../git_subtrees/cva6/src/pmp/src/pmp.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpnew_opgroup_fmt_slice.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpnew_divsqrt_multi.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpnew_fma_multi.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpnew_classifier.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpnew_noncomp.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpnew_cast_multi.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpnew_opgroup_block.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpnew_rounding.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpnew_top.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpnew_fma.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpnew_opgroup_multifmt_slice.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpu_div_sqrt_mvp/hdl/iteration_div_sqrt_mvp.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpu_div_sqrt_mvp/hdl/nrbd_nrsc_mvp.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpu_div_sqrt_mvp/hdl/div_sqrt_top_mvp.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpu_div_sqrt_mvp/hdl/preprocess_mvp.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpu_div_sqrt_mvp/hdl/control_mvp.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpu_div_sqrt_mvp/hdl/norm_div_sqrt_mvp.sv \
               ../../../git_subtrees/cva6/src/fpu/src/fpu_div_sqrt_mvp/hdl/div_sqrt_mvp_wrapper.sv \
               ../../../git_subtrees/cva6/src/frontend/frontend.sv \
               ../../../git_subtrees/cva6/src/frontend/instr_scan.sv \
[... 100 lines cut ...]
               ../../../git_subtrees/cva6/src/common_cells/src/shift_reg.sv \
               ../../../git_subtrees/cva6/src/common_cells/src/exp_backoff.sv \
               ../../../git_subtrees/cva6/src/scoreboard.sv \
               ../../../git_subtrees/cva6/src/store_buffer.sv \
               ../../../git_subtrees/cva6/src/store_unit.sv \
               ../../../git_subtrees/cva6/src/axi_adapter.sv \
               ../../../git_subtrees/cva6/src/fpu_wrap.sv \
               ../../../git_subtrees/cva6/src/tlb.sv \
               ../../../git_subtrees/cva6/src/ariane_regfile_ff.sv \
[... cut ...]
               ../../../git_subtrees/cva6/src/register_interface/register_interface/src/apb_to_reg.sv \
               ../../../git_subtrees/cva6/fpga/src/apb_timer/apb_timer.sv \
               ../../../git_subtrees/cva6/src/axi/src/axi_multicut.sv \
               ../../../git_subtrees/cva6/src/tech_cells_generic/src/cluster_clock_inverter.sv \
               ../../../git_subtrees/cva6/src/tech_cells_generic/src/pulp_clock_mux2.sv \
               ../../../git_subtrees/cva6/fpga/src/apb_timer/timer.sv
prj_set_impl_opt -impl impl1 top ariane
prj_run_synthesis

MikeOpenHWGroup, 2023-02-17:
Hi @Saqlinahamad, since this issue has not been updated in ~1.5 years, I will close it.

closed 2023-02-17 by MikeOpenHWGroup