openhwfoundation/cva6 #741: Need of a NOP instruction while running CSR test program in Verilator

RanjanThales, 2021-09-30:
Hi,
While working on csr test program, I am seeing first instruction is not getting considered by Verilator simulation. This is leading to first line mis-match with Spike model simulation.
[image]

Note that: In case, I put a NOP instruction at the start of test program, I could resolve this particular issue.

-Regards

andreaskuster, 2022-01-05:
I assume there is an underlying problem present. However, adding a nop instruction in your C code could be achieved with the following line: `asm volatile ("nop" : : );` or `asm volatile ("addi x0, x1, 0" : : );`

MikeOpenHWGroup, 2022-04-12:
Agree with @andreaskuster, that there is an underlying problem.  Adding a nop as the first instruction is _not_ a viable solution nor is it a work-around we should be satisfied with.

I would start by dumping waves and tracing the first instruction from the AXI interface through to the RVFI outputs.  If the first instruction appears at the RVFI, then look at the tracer itself.   There is a simple (well, hopefully simple) bug in either the RTL or Testbench that is causing this issue.

RanjanThales, 2022-05-10:
Hi Mike,
As discussed earlier on this, we have a similar issue while working on the riscv-arch-testsuit in cva6 environment. The conversion scripts seems like some issue i.e. verilator conversion script. This script is not writing executed instruction at 0x8000_0000 from .log file into .csv one. I have tested also tested for riscv-arch-tests , its also working fine. For reference: https://github.com/openhwgroup/core-v-verif/pull/1254.
So I believe we can close this issue now.

closed 2022-05-10 by RanjanThales