openhwfoundation/cva6 #744: Consolidate top-level CVA6 parameters into struct of parameters

MikeOpenHWGroup, 2021-10-01:
This issue is related to #743.

The [ariane_cfg_t](https://github.com/openhwgroup/cva6/blob/master/core/include/ariane_pkg.sv#L36) struct does not "contain" all the parameters needed to fully define all CVA6 configurations.  (There is even a [TODO](https://github.com/openhwgroup/cva6/blob/master/core/include/ariane_pkg.sv#L130) comment in the code about it.)

This issue can be closed when _all_ parameters required to implement any of the "golden configurations" mentioned in #743 are added to `ariane_cfg_t`.

MikeOpenHWGroup, 2023-08-28:
Hi @JeanRochCoulon.  I believe this Issue can be closed now.  Let me know if you agree and I will do it.

JeanRochCoulon, 2023-08-28:
This issue will be closed when #1321 will be merged. As this PR is in process, done by @zarubaf, I re-assign to him the issue

zarubaf, 2023-10-23:
#1321 is merged

closed 2023-10-23 by zarubaf