openhwfoundation/cva6 #745: Document parameters used to implement CVA6 configurations

MikeOpenHWGroup, 2021-10-01:
This issue is related to #743 and #744, and exists to ensure that the golden configurations and their associated parameters in [ariane_cfg_t](https://github.com/openhwgroup/cva6/blob/master/core/include/ariane_pkg.sv#L36) are described in the CVA6 User Manual.  A good example of this can be found in the [Integration](https://cv32e40p.readthedocs.io/en/latest/integration/#parameters) chapter of the CV32E40P user manual.

MikeOpenHWGroup, 2023-02-17:
This issue needs to be updated for CV32A6_v5.0.0.

JeanRochCoulon, 2023-03-14:
The CVA6 parameters and the cv32a6_v5.0.0 configuration are described in this user guide chapter https://docs.openhwgroup.org/projects/cva6-user-manual/01_cva6_user/Parameters_Configuration.html