openhwfoundation/cva6 #747: Combinational loops involving DRAM signals

eldrickm, 2021-10-01:
I am using Questa Version 10.5 to compile / elaborate the design for formal verification.

I am trying to setup the cva6 core in a test harness which only has the DRAM and no other peripherals for formal verification, with the instruction signal being a cutpoint.

I have attached the top level harness file I am using which instantiates an unmodified ariane.sv module and the DRAM system. It is essentially the same as ariane_testharness.sv with unneeded peripherals commented out.

I am getting an issue where the tool is finding 3 combinational loops in the design. Relevant snippets below, full formal_design.rpt attached as well:
```
Type      : COMBO_LOOP
Loop Signal: i_axi_riscv_atomics.mst.ar_valid
Loop Signal: dram.ar_valid
Loop Signal: i_axi_delayer.ar_valid_i
Loop Signal: i_axi_delayer.i_stream_delay_ar.valid_i
Loop Signal: i_axi_delayer.i_stream_delay_ar.valid_o
Loop Signal: i_axi_delayer.ar_valid_o
Loop Signal: dram_delayed.ar_valid
Loop Signal: i_axi2mem.slave.ar_valid
Loop Signal: i_axi2mem.slave.w_ready
Loop Signal: dram_delayed.w_ready
Loop Signal: i_axi_delayer.w_ready_i
Loop Signal: i_axi_delayer.i_stream_delay_w.ready_i
Loop Signal: i_axi_delayer.i_stream_delay_w.ready_o
Loop Signal: i_axi_delayer.w_ready_o
Loop Signal: dram.w_ready
Loop Signal: i_axi_riscv_atomics.mst.w_ready
Loop Signal: i_axi_riscv_atomics.i_atomics.mst_w_ready_i
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.mst_w_ready_i
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.wr_clr_req
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.inp_valid_i
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.i_arb.inp_valid_i
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.i_arb.gen_rr_arb.i_arbiter.req_i
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.i_arb.gen_rr_arb.i_arbiter.genblk1.req_d
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.i_arb.gen_rr_arb.i_arbiter.genblk1.req_nodes
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.i_arb.gen_rr_arb.i_arbiter.req_o
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.i_arb.oup_valid_o
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.oup_valid_o
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_art.clr_req_i
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_art.set_gnt_o
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.mst_ar_valid_o
Loop Signal: i_axi_riscv_atomics.i_atomics.mst_ar_valid_o
```

```
Type      : COMBO_LOOP
Loop Signal: i_axi_riscv_atomics.mst.aw_valid
Loop Signal: dram.aw_valid
Loop Signal: i_axi_delayer.aw_valid_i
Loop Signal: i_axi_delayer.i_stream_delay_aw.valid_i
Loop Signal: i_axi_delayer.i_stream_delay_aw.valid_o
Loop Signal: i_axi_delayer.aw_valid_o
Loop Signal: dram_delayed.aw_valid
Loop Signal: i_axi2mem.slave.aw_valid
Loop Signal: i_axi2mem.slave.w_ready
Loop Signal: dram_delayed.w_ready
Loop Signal: i_axi_delayer.w_ready_i
Loop Signal: i_axi_delayer.i_stream_delay_w.ready_i
Loop Signal: i_axi_delayer.i_stream_delay_w.ready_o
Loop Signal: i_axi_delayer.w_ready_o
Loop Signal: dram.w_ready
Loop Signal: i_axi_riscv_atomics.mst.w_ready
Loop Signal: i_axi_riscv_atomics.i_atomics.mst_w_ready_i
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.mst_w_ready_i
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.wr_clr_req
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.inp_valid_i
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.i_arb.inp_valid_i
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.i_arb.gen_rr_arb.i_arbiter.req_i
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.i_arb.gen_rr_arb.i_arbiter.genblk1.req_d
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.i_arb.gen_rr_arb.i_arbiter.genblk1.req_nodes
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.i_arb.gen_rr_arb.i_arbiter.req_o
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.i_arb.oup_valid_o
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_non_excl_acc_arb.oup_valid_o
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_art.clr_req_i
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.i_art.check_gnt_o
Loop Signal: i_axi_riscv_atomics.i_atomics.i_lrsc.mst_aw_valid_o
Loop Signal: i_axi_riscv_atomics.i_atomics.mst_aw_valid_o
```

```
Type      : COMBO_LOOP
Loop Signal: i_axi_master_connect_ariane.master.r_ready
Loop Signal: slave[0].r_ready
Loop Signal: i_axi_xbar.slave[0].r_ready
Loop Signal: i_axi_xbar.s_slave_r_ready
Loop Signal: i_axi_xbar.axi_node_i.slave_rready_i
Loop Signal: i_axi_xbar.axi_node_i._RESP_BLOCK_GEN[0].RESP_BLOCK.rready_i
Loop Signal: i_axi_xbar.axi_node_i._RESP_BLOCK_GEN[0].RESP_BLOCK.BR_ALLOC.rready_i
Loop Signal: i_axi_xbar.axi_node_i._RESP_BLOCK_GEN[0].RESP_BLOCK.BR_ALLOC.error_gnt_o
Loop Signal: i_axi_xbar.axi_node_i._RESP_BLOCK_GEN[0].RESP_BLOCK.AR_ADDR_DEC.error_gnt_i
Loop Signal: i_axi_xbar.axi_node_i._RESP_BLOCK_GEN[0].RESP_BLOCK.AR_ADDR_DEC.arready_int
Loop Signal: i_axi_xbar.axi_node_i._RESP_BLOCK_GEN[0].RESP_BLOCK.AR_ADDR_DEC.arready_o
Loop Signal: i_axi_xbar.axi_node_i._RESP_BLOCK_GEN[0].RESP_BLOCK.arready_o
Loop Signal: i_ariane.i_cache_subsystem.i_adapter.i_axi_shim.rd_gnt_o
Loop Signal: i_ariane.i_cache_subsystem.i_adapter.arb_gnt
Loop Signal: i_ariane.i_cache_subsystem.i_adapter.invalidate
Loop Signal: i_ariane.i_cache_subsystem.i_adapter.axi_rd_rdy
```

In the first two cases, the combinational loop occurs in axi2mem.sv, where the slave.w_ready signal is sensitized to the value of slave.ar_valid and slave.aw_valid respectively. slave.w_ready is then routed through the atomics module which has ar_valid and aw_valid sensitized to w_ready, creating a loop.

In the third case, a similar event occurs between slave[0].r_ready and the error_gnt_o signal.

Has anyone else ran into this issue? How should I resolve it?

[ariane_testharness.sv.txt](https://github.com/openhwgroup/cva6/files/7269931/ariane_testharness.sv.txt)
[formal_design.rpt.txt](https://github.com/openhwgroup/cva6/files/7269932/formal_design.rpt.txt)

MikeOpenHWGroup, 2023-02-17:
Hi @JeanRochCoulon and @ASintzoff.  This issue is very old, but it goods "interesting".  Could you assign someone to investigate?

JeanRochCoulon, 2023-03-16:
Hello @MikeOpenHWGroup, this seems to be related to APU and not CVA6 itself. It would be easy to reproduce having the synthesis flow functional at testharness level. As it is not the case Today, I re-assign it to you. To my mind, we can close it.

MikeOpenHWGroup, 2023-03-23:
Agreed @JeanRochCoulon.  Closing as "won't fix".

closed 2023-03-23 by MikeOpenHWGroup