openhwfoundation/cva6 #782: make verilate failed.

sreersr, 2022-01-03:
I was trying to use RISCV toolchain with the Ariane core from https://github.com/openhwgroup/cva6 in macbook pro, I used homebrew to build riscv-toolchain and trying to build it, but while running **_make Verilate_** the following error is observed, please advise. it will be appreciated if you give separate advise for my queries. Thanks in advance.

[image]
[image]

andreaskuster, 2022-01-04:
The missing headers `fesvr/dtm.h` respectively `fesvr/dtm.h` are both part of https://github.com/riscv/riscv-isa-sim.git I suggest to install it either manually or using the [ci script](https://github.com/openhwgroup/cva6/blob/master/ci/install-fesvr.sh) and try again.

sreersr, 2022-01-05:
> The missing header `fesvr/dtm.h` are both part of https://github.com/riscv/riscv-isa-sim.git I suggest to install it either manually or using the [ci script](https://github.com/openhwgroup/cva6/blob/master/ci/install-fesvr.sh) and try again.

Thanks Andreaskuster, can you help me to know how to run this ci script in MacBook Pro, FYI I have installed RISC-V tool using home-brew and my terminal is zsh. Please advise. Thanks in advance.

andreaskuster, 2022-01-05:
I suggest do use the `bash` shell instead of `zsh`. What error are you running into by executing the provided script above?

Furthermore, these scripts are known to work on `ubuntu 20.04`. I highly suggest to start with a simple setup (i.e. ubuntu 20.04 in on another system or a virtual machine) and advancing to macOS later on.

MikeOpenHWGroup, 2023-02-17:
Hi @SreeRanjaniRajendran, what is the current status of this Issue?  There has not been a comment for more than one year, so if we do not hear from you by 2023-03-02 we will close the issue.   Thanks for your interest in CVA6!

MikeOpenHWGroup, 2023-03-13:
Closing due to a lack of activity on this issue.

closed 2023-03-13 by MikeOpenHWGroup