openhwfoundation/cva6 #794: File list for CVA6 core?

KeithoHimself, 2022-01-17:
I am attempting to instantiate a CVA6 into a SoC.  Looking through the source tree, I cannot seem to find a single file list which provides a list of only the source files needed for the CVA6 core.  Am I missing something?

zarubaf, 2022-01-18:
The core standalone flists are in `core` i.e., https://github.com/openhwgroup/cva6/blob/master/core/Flist.cv64a6_imafdc_sv39

Hope that helps.

KeithoHimself, 2022-01-18:
Yes, I saw that file list and that is the one I used in my environment.

It appears that the file list is incorrect or incomplete.  Compiling with Questa 2019.1 using the file list fails (partial transcript below).

Steps to reproduce:
1. export CVA6_REPO_DIR=<path to cva6 repo>
2. vlog -f Flist.cv64a6_imafdc_sv39

Here is  partial transcript:* Warning: /tmp/cva6/common/submodules/common_cells/src/lfsr.sv(262): (vlog-2583) [SVCHK] - Some checking for conflicts with always_comb and always_latch variables not yet supported. Run vopt to provide additional design-level checks.
-- Compiling module lzc
** Error: /tmp/cva6/common/submodules/common_cells/src/lzc.sv(33): (vlog-2164) Class or package 'cf_math_pkg' not found.
** Error: /tmp/cva6/common/submodules/common_cells/src/lzc.sv(33): (vlog-2730) Undefined variable: 'cf_math_pkg'.
** Error: (vlog-13069) /tmp/cva6/common/submodules/common_cells/src/lzc.sv(33): near "::": syntax error, unexpected ::, expecting ')'.
-- Compiling module rr_arb_tree

...

-- Compiling module sram
-- Compiling module mmu
-- Compiling module ptw
-- Compiling module tlb
End time: 12:17:23 on Jan 18,2022, Elapsed time: 0:00:00
Errors: 816, Warnings: 6

BTW, there are two cf_math_pkg.sv source files in the project.

Looking at the Makefile, it does not appear that the file list is used for compilation.  Doing an "ack" to find a use or reference to the file list "Flist.cv64a6_imafdc_sv39" does result in any hits..

It looks like the file list is not being used within the project to specify core source files - and the file list is out of date.

Wanted to post the issue to understand whether the file list is "truth" for the core source files.

zarubaf, 2022-01-19:
It might be that @Gchauvon just fixed your issue (966b6b3e), is that possible?

Yes, the Makefile is legacy, but the `flist`s are used in `core-v-verif`.

KeithoHimself, 2022-01-21:
Yes, that fixed the issue, thanks!

I'd recommend adopting a "correct by construction" approach that uses the file lists in the makefile as well as in all other relevant areas.

closed 2022-01-21 by KeithoHimself