openhwfoundation/cva6 #800: Questions regarding simulation with Questa

KeithoHimself, 2022-01-22:
I am working to get the simulations described in the README running under Questa and I am running into issues.  Before digging deeper, I'd like to know the following:

1) Is the flow in the ./ci directory currently running under Questa?  Is this flow run nightly?

2) What version of Questa is being used?
    - I have found that modern version of Questa throw a sig 11 (i.e. they crash) when loading the simulation
    - Questa 2017.3 will load and run the sim, but when running "hello world", I do not see "Hello World!" in the transcript.
 
3) When using Questa, the sim tries to load libfesvr.so.  Where is that library being built?
   - The ./ci/install-fesvr recipe does not appear to build and install libfesvr.so, only libfesvr.a
   - A manual change to riscv-isa-sim/fesv/esvr.mk.in to add "fesvr_install_shared_lib = yes" appears to be required

rickoco, 2022-01-23:
Hey @KeithoHimself thanks for your comments.  Within the OpenHW Group compute platform on AWS, we're in the process of setting up and running various CI flows using Cadence and Synopsys tools and OpenHW community members also have Siemens / Mentor based flows.  It would be good to learn more about your environment.  Email me at rickoco@openhwgroup.org so we can setup a time to chat.
Rick

christian-lanius, 2022-02-22:
@KeithoHimself 
I am having the same issue, i.e. I am unable to simulate the design at this point in time using Questasim.
I have tried two versions of questasim: 
2018-19 Europractice version results in an error 
```
Error: {PATH}/cva6/corev_apu/clint/axi_lite_interface.sv(55): Internal error: unhandled IDL type ../../src/vlog/vsymtab.c(5443) <net_type>. Please contact Questa support at http://supportnet.mentor.com/
```
If it helps, I can provide the full list of these errors. 

Running the 2020-21 EP version results in the same sig 11 error you got. As the issue occurs in the opt step, I can circumvent it by skipping this step: Commenting out `$(VOPT) $(compile_flag) -work $(library)  $(top_level) -o $(top_level)_optimized +acc -check_synthesis` (line 344), replacing the name in the sim command in the Makefile with the unoptimized version and launching simulation with `QUESTASIM_FLAGS="-novopt -suppress 12110"` makes the simulation start. I also have to set the MODELSIM_HOME variable. However, the simulation then still crashes because of an infinite loop detected. 

So, even getting rid of the sig 11 error is not enough to get the simulation to run for me.

Did you make any progress in this issue you could share with me?

davidmallasen, 2022-03-29:
@christian-lanius
What do you mean by "replacing the name in the sim command in the Makefile with the unoptimized version"? I am trying to follow your steps.

christian-lanius, 2022-03-29:
I have attached a patch which contains the changes I made to the Makefile to get the different error message
[makefile.txt](https://github.com/openhwgroup/cva6/files/8370922/makefile.txt)
. You can then launch your simulation like this:
```
make sim elf-bin=BIN_TO_EXECUTE QUESTASIM_FLAGS="-novopt -suppress 12110"
```

christian-lanius, 2022-03-29:
I have further debugged this issue and found the pullrequest in the https://github.com/pulp-platform/riscv-dbg/pull/102. The issue that pull request solves sounded suspiciously like the issue I am facing when I run ariane without optimization (as described above). Applying a similar patch to the dm_mem.sv file ([attached](https://github.com/openhwgroup/cva6/files/8371732/dm_mem.txt)), fixes the infinite loop I was getting.
Unfortunately, this does not fix the issue of the signal 11 error we see when we optimize the design, but it at least allows us to simulate the latest master branch.
The patch does not change any logic, it only gets rid of the default assign.

MikeOpenHWGroup, 2023-02-11:
@christian-lanius, thanks for your support on this issue.  Can you confirm the current status?  It seems that a solution was identified (PR #988), but is not yet merged in.  Is that correct?

christian-lanius, 2023-02-13:
Yes, the work that @niwis was doing to fix the scripts and the more recent versions of questasim do fix the issue, as far as I can tell.
I am unsure if the shared object issue has been fixed, as I haven't bothered to build fully from scratch in some time.

JeanRochCoulon, 2023-10-24:
Related to #1569

JeanRochCoulon, 2023-12-07:
As this issue thread is no more active, and related to Questasim which is not supported. I close it.

closed 2023-12-07 by JeanRochCoulon