openhwfoundation/cva6 #809: common_cells folder location in cva6 repo

nathpranay, 2022-02-02:
In current folder structure of cva6 repo "common_cells" folder has been placed in "cva6\core\fpu\src\" . Will it be a good idea to place "common_cells" inside "cva6\core".

MikeOpenHWGroup, 2022-02-02:
Hi @nathpranay, thanks for the question.  I have looked into this and I think everything is OK. As you may know, both the fpu and common_cells are brought into this repo as [git submodules](https://github.blog/2016-02-01-working-with-submodules/).  Both the fpu and common_cells are maintained by the PULP team on their repos.

As it turns out, the cva6 uses one set of files from common_cells and these are found in `cva6/common/submodules/common_cells` and the fpu also uses some files from common_cells and these are found in `cva6/core/fpu/src/common_cells`.

The confusing bit is that the fpu is a submodule for cva6 and common_cells is a submodule for both cva6 and the fpu.  So when we recursively clone the cva6 we get a clone of the fpu into `core/fpu` which in turn clones common_cells into `cva6/core/fpu/src/common_cells`.  As it turns out, the CVA6 manifest does not reverse any of the common_cells from that path, so I think we are fine.

@zarubaf, can you confirm that I have this right?

nathpranay, 2022-02-02:
Thanks Mike. I think we can close this issue.

MikeOpenHWGroup, 2023-02-17:
Closing.  ...one year late...

closed 2023-02-17 by MikeOpenHWGroup