openhwfoundation/cva6 #811: no makefile in core/example_tb

Rayz-b, 2022-02-04:
error while running following code from README.md:
<CODE>
$ cd core/example_tb
$ make veri_run
</CODE>

Rayz-b, 2022-02-04:
ALSO: <code>make help</code> doesn't work
<code>make: *** no rule to make target help</code>

zarubaf, 2022-02-04:
@MikeOpenHWGroup what was the intention of `example_tb`?

TNdu, 2022-02-15:
I too have this error and cannot run the "make veri_run" command as stated in the **Build Model and Run Simulation** section at **To build, compile and run the CVA6 core-only in its example testbench using Verilator (known to work with V4.108)** step.

zarubaf, 2022-05-04:
Ping @MikeOpenHWGroup

pjlobo, 2023-01-16:
Please see https://github.com/openhwgroup/cva6/issues/877#issuecomment-1384234340

MikeOpenHWGroup, 2023-02-17:
My apologies for letting this Issue go stale.  The `example_tb`was a failed experiment from more than a year ago.   This issue can be closed.

closed 2023-02-17 by MikeOpenHWGroup

hhhsiang, 2023-03-01:
Hi @MikeOpenHWGroup ,so the commands:
`$ cd core/example_tb`
`$ make veri_run`
are meaningless now, right? Thanks.