openhwfoundation/cva6 #812: ../configure: No such file or directory

Rayz-b, 2022-02-04:
error while running following code from [README.md](https://github.com/openhwgroup/cva6#running-user-space-applications):
<code>mkdir build</code>
<code>cd build</code>
<code>../configure --prefix=$RISCV --host=riscv64-unknown-elf</code>
ERROR:
<code>../configure: No such file or directory</code>
the <code>configure</code> file seems to be missing from root directory of the project

pcotret, 2022-02-17:
These commands are basically done in this script: https://github.com/openhwgroup/cva6/blob/master/ci/install-riscvpk.sh
It will `cd` in the `riscv-pk`repository. The `configure`file is available there: https://github.com/riscv-software-src/riscv-pk/blob/master/configure

MikeOpenHWGroup, 2023-02-17:
Thanks for your help with this @pcotret.

Hi @Rayz-b.  This Issue was responded to a year ago so I am going to assume that the problem is resolved and will Close the Issue.  Please feel free to re-open it (or create a new one) if you are still having problems.

closed 2023-02-17 by MikeOpenHWGroup

hhhsiang, 2023-03-01:
hi @pcotret , so we don't need to execute the following commands, right?
`mkdir build`
`cd build`
`../configure --prefix=$RISCV --host=riscv64-unknown-elf`
`make`
`make install`
They have been done in this script: https://github.com/openhwgroup/cva6/blob/master/ci/install-riscvpk.sh
Thank you very much!