openhwfoundation/cva6 #824: make fpga error *.dcp file not existing

TNdu, 2022-02-15:
Hello,

I am trying to run the "make fpga" command after the successfull "make verilate" command and i'm getting this error :+1: 

1 Infos, 0 Warnings, 0 Critical Warnings and 1 Errors encountered.
synth_design failed
ERROR: [Runs 36-527] DCP does not exist: /home/tnieddu/PFE/Vivado/CLEAN_REPO/cva6/corev_apu/fpga/xilinx/xlnx_axi_dwidth_converter_dm_master/xlnx_axi_dwidth_converter_dm_master.gen/sources_1/ip/xlnx_axi_dwidth_converter_dm_master/xlnx_axi_dwidth_converter_dm_master.dcp
INFO: [Common 17-206] Exiting Vivado at Tue Feb 15 11:47:43 2022...
make[1]: *** [Makefile:28: work-fpga/ariane_xilinx.bit] Error 1
make[1]: Leaving directory '/home/tnieddu/PFE/Vivado/CLEAN_REPO/cva6/corev_apu/fpga'
make: *** [Makefile:783: fpga] Error 2

As i don't even know what those *.dcp files are, i am stuck and can't find any information elsewhere hence why i am opening this issue.

I am trying to get the .xpr to open the project on vivado and try to enhance the RISCV core for academic project purposes.

Could you please help me ?

TNdu, 2022-02-17:
I found a way to counter this :+1: 
I opened the arian.xpr in Vivado and did "Generate Output Products..." with the Out Of Context per IP option selected.
As the missing *.dcp file is generated, i can run the "make fpga" command until it's finished correctly.

However now, i have the same problem as #651 , the Vivado project is empty.

As i stated before, i'd like to open the CVA6 project in Vivado to try and enhance the RISCV core.

Could you please help ?

Gchauvon, 2022-02-18:
After this line : 
https://github.com/openhwgroup/cva6/blob/e4b48a794b0229eeb9264a6b3f459c86545d1d85/corev_apu/fpga/scripts/run.tcl#L72
You can save the project to open it in GUI mode by adding this to the script :
```
#save_project_as ariane_gui
#exit 1
```

You can then open ariane_gui.xpr in Vivado and do whatever you like.

MikeOpenHWGroup, 2023-02-17:
Hi @TNdu.  You responded with a "thumbs-up" one year ago, so I am going to assume this Issue was resolved.

Thanks for helping out @Gchauvon.

Closing.

closed 2023-02-17 by MikeOpenHWGroup