openhwfoundation/cva6 #827: MMU should not be present when using config cv32a6_imac_sv0

Gchauvon, 2022-02-18:
In `ariane_pkg.sv` parameter `MMU_PRESENT` is always set. It should not be the case since cv32a6_imac_sv0 configuration does not 
 have MMU.

https://github.com/openhwgroup/cva6/blob/e4b48a794b0229eeb9264a6b3f459c86545d1d85/core/include/ariane_pkg.sv#L625

I recommend to make this parameter depends on which cva6 config package we use. 

- ` cv32a6_imac_sv0_config_pkg.sv`
-  `cv32a6_imac_sv32_config_pkg.sv`
- `cv32a6_imafc_sv32_config_pkg.sv`
- `cv64a6_imafc_sv39_config_pkg.sv`

I believe we should also modify parameter like SV_MODE, SV, VLEN ? accordingly.
https://github.com/openhwgroup/cva6/blob/e4b48a794b0229eeb9264a6b3f459c86545d1d85/core/include/riscv_pkg.sv#L48-L49

Some more RTL may or may not need changes if MMU is disabled but I didn't look into that very deeply.

MikeOpenHWGroup, 2023-02-17:
Hi @Gchauvon.  This Issue is one year old and has not received any updates.   If it is no longer valid, please change the `Status:New` label to `Status:Resolved` or `Status:Wont-fix` and close it.  If the issue is still valid, please work to implement a fix or assign it to someone else.

Gchauvon, 2023-12-06:
Issue is not relevant anymore since parametrization.

closed 2023-12-06 by Gchauvon