openhwfoundation/cva6 #828: Error running make verilate DROMAJO=1 -reg

sreersr, 2022-02-22:
Hi I am trying to use Dromajo to do HW/SW co simulation of cva6. As of now, I can simulate ariane with verilator simulator. Now I have to use Dromajo and while running make verilate DROMAJO=1, I got below error.
../corev_apu/tb/ariane_tb.cpp: In function ‘int main(int, char**)’:
../corev_apu/tb/ariane_tb.cpp:323:49: error: ‘class Variane_testharness’ has no member named ‘ariane_testharness__DOT__i_sram__DOT__gen_cut__BRA__0__KET____DOT__gen_mem__DOT__i_ram__DOT__Mem_DP’; did you mean ‘ariane_testharness__DOT__i_sram__DOT__gen_cut__BRA__0__KET____DOT__gen_dromajo__DOT__i_ram__DOT__unnamedblk1__DOT__f_name’?
  323 |   memif.read(0x80000000, mem_size, (void *)top->ariane_testharness__DOT__i_sram__DOT__gen_cut__BRA__0__KET____DOT__gen_mem__DOT__i_ram__DOT__Mem_DP);
      |                                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                                 ariane_testharness__DOT__i_sram__DOT__gen_cut__BRA__0__KET____DOT__gen_dromajo__DOT__i_ram__DOT__unnamedblk1__DOT__f_name
make[1]: *** [Variane_testharness.mk:69: ariane_tb.o] Error 1
make[1]: *** Waiting for unfinished jobs....
^Z
[1]+  Stopped                 make verilate DROMAJO=1

Please advise to fix this issue. My Verilator version is Verilator 4.100 2020-09-07 rev UNKNOWN_REV. Thanks in advance.

kabylkas, 2022-02-22:
Hi 👋 I was responsible for the integration of Dromajo :) The CVA6 codebase has evolved since I integrated Dromajo and I feel like some changes could have broken the Dromajo pass. There a lot of things going on in my life currently, so I will need some time before I can look into this.

sreersr, 2022-03-01:
Hey, thanks for your reply. Hope you will be back to normal soon. Let me know once you are ready to fix this issue.

MikeOpenHWGroup, 2023-02-17:
Hi @kabylkas and @SreeRanjaniRajendran.  This Issue has not been updated in almost one year.  Is it still relevant?  If we have not heard from you by 2023-03-02, we will Close the Issue.  Thanks.

MikeOpenHWGroup, 2023-03-13:
Closing due to a lack of activity on this issue.

closed 2023-03-13 by MikeOpenHWGroup