openhwfoundation/cva6 #829: Debug Specification for CVA6

MikeOpenHWGroup, 2022-02-25:
According to their respective User Manuals, both the CV32E40X and CV32E40S have adopted Version1.0.0-STABLE of the RISC-V Debug Support specification.  As this has not yet been ratified by RISC-V International, the current version of the CVA6 specification points to version 0.13.2 as this is the most up-to-date _ratified_ version of the RISC-V Debug spec.

There is always a natural tension between the desire to be as up-to-date as possible and the need to maintain compliance to a ratified spec.   @Silabs-ArjanB, can you comment on why the CV32E40X/S decided to go with v1.0.0?   Thanks.

Silabs-ArjanB, 2022-02-28:
The CV32E40X/CV32E40S is targeting various not yet ratified RISC-V specifications. With respect to Debug we are going to implement various new features that were not present in the CV32E40P. That we could have done using the 0.13.2 spec as well, but we simply want to be in sync with the latest spec. The 1.0 spec is also clarifying various things that were ambiguous or not well defined in the 0.13.2 spec.

MikeOpenHWGroup, 2023-02-17:
Thanks @Silabs-ArjanB.

closed 2023-02-17 by MikeOpenHWGroup