openhwfoundation/cva6 #837: fpu_valid_o in issue_stage.sv

misomochi, 2022-03-11:
I was learning about how Fpnew is integrated into the Ariane core and wanted to make sure how its handshake interface (ready/valid protocol) is being dealt with. When I traced the Fpnew's ready/valid signals from bottom-up, I realized that the valid signal was supposed to be generated in the issue stage. However, no wires or registers are assigned to fpu_valid_o within the issue_stage module. This should cause fpu_valid_o being undefined and hence affecting the functionality of Fpnew? Or did I misunderstand sth?

zarubaf, 2022-03-11:
It is driven from a registe: https://github.com/openhwgroup/cva6/blob/e197b445fc8f19dae8ce9ac0493396ce2bc19480/core/issue_read_operands.sv#L126

misomochi, 2022-03-11:
Hi @zarubaf , thanks for answering! I totally overlooked that module and I apologize for that. However, the _issue_read_operands_ module instantiated in issue_stage.sv does not have the handshake ports for the floating point unit:
https://github.com/openhwgroup/cva6/blob/e197b445fc8f19dae8ce9ac0493396ce2bc19480/core/issue_stage.sv#L160-L191
Does this imply the Fpnew is being disabled temporarily? Sorry if I overlooked any details or Systemverilog syntaxes, I'm more familiar with Verilog.

davidmallasen, 2022-03-11:
The .* at the end means that the ports are connected to the variables of the same name. I think they are called implicit port connections.

misomochi, 2022-03-13:
Ahh no wonder I could find the signal line with Ctrl+F. Thanks for clarifying @davidmallasen !

closed 2022-03-13 by misomochi