openhwfoundation/cva6 #842: Segmentation fault using QuestaSim

davidmallasen, 2022-03-23:
Hello,

The latest version of the repo (8d893fb647d758bed5876307c64caa6adbc89f97) triggers a segmentation fault when trying to simulate using QuestaSim. This does not happen with commit 00236be3d8552f93a0bebda8f9820ec54b64a000 . The steps I have followed are:
~~~
git clone https://github.com/openhwgroup/cva6.git
cd cva6
git submodule update --init --recursive

mkdir tmp

echo '
#include <stdio.h>

int main(int argc, char const *argv[]) {
    printf("Hello CVA6!\\n");
    return 0;
}' > hello.c

riscv64-unknown-elf-gcc hello.c -o hello.elf

make sim elf-bin=$RISCV/riscv64-unknown-elf/bin/pk target-options=hello.elf batch-mode=1
~~~

This outputs the following:
~~~
Makefile:151: XCELIUM_HOME not set which is necessary for compiling DPIs when using XCELIUM
vlib work
vlog +cover=bcfst+/dut -incr -64 -nologo -quiet -suppress 13262 -permissive +define+WT_DCACHE+RVFI_TRACE -work work /home/david/cva6/core/include/cv64a6_imafdc_sv39_config_pkg.sv /home/david/cva6/core/include/riscv_pkg.sv /home/david/cva6/corev_apu/riscv-dbg/src/dm_pkg.sv /home/david/cva6/core/include/ariane_pkg.sv /home/david/cva6/core/include/ariane_rvfi_pkg.sv /home/david/cva6/core/include/wt_cache_pkg.sv /home/david/cva6/core/include/cvxif_pkg.sv /home/david/cva6/corev_apu/axi/src/axi_pkg.sv /home/david/cva6/corev_apu/register_interface/src/reg_intf.sv /home/david/cva6/core/include/axi_intf.sv /home/david/cva6/corev_apu/tb/rvfi_pkg.sv /home/david/cva6/corev_apu/tb/ariane_soc_pkg.sv /home/david/cva6/corev_apu/tb/ariane_axi_soc_pkg.sv /home/david/cva6/core/include/ariane_axi_pkg.sv /home/david/cva6/core/include/std_cache_pkg.sv /home/david/cva6/core/fpu/src/fpnew_pkg.sv /home/david/cva6/common/submodules/common_cells/src/cf_math_pkg.sv /home/david/cva6/core/cvxif_example/include/cvxif_instr_pkg.sv /home/david/cva6/core/fpu/src/fpu_div_sqrt_mvp/hdl/defs_div_sqrt_mvp.sv  +incdir+common/submodules/common_cells/include/  +incdir+corev_apu/axi/include/  +incdir+corev_apu/register_interface/include/ -suppress 2583
** Warning: (vlog-13288) Multiple macros defined in +define+ command line switch.
# vcom -64 -nologo -quiet -2008 -work work -pedanticerrors 
vlog +cover=bcfst+/dut -incr -64 -nologo -quiet -suppress 13262 -permissive +define+WT_DCACHE+RVFI_TRACE -timescale "1ns / 1ns" -work work /home/david/cva6/core/include/instr_tracer_pkg.sv /home/david/cva6/common/local/util/instr_tracer_if.sv /home/david/cva6/common/local/util/instr_tracer.sv /home/david/cva6/corev_apu/src/tech_cells_generic/src/cluster_clock_gating.sv /home/david/cva6/corev_apu/tb/common/mock_uart.sv /home/david/cva6/common/local/util/sram.sv  +incdir+common/submodules/common_cells/include/  +incdir+corev_apu/axi/include/  +incdir+corev_apu/register_interface/include/ -suppress 2583
** Warning: (vlog-13288) Multiple macros defined in +define+ command line switch.
# Suppress message that always_latch may not be checked thoroughly by QuestaSim.
vcom -64 -nologo -quiet -2008 -work work -pedanticerrors /home/david/cva6/corev_apu/fpga/src/apb_uart/src/apb_uart.vhd /home/david/cva6/corev_apu/fpga/src/apb_uart/src/uart_transmitter.vhd /home/david/cva6/corev_apu/fpga/src/apb_uart/src/uart_interrupt.vhd /home/david/cva6/corev_apu/fpga/src/apb_uart/src/slib_mv_filter.vhd /home/david/cva6/corev_apu/fpga/src/apb_uart/src/slib_input_filter.vhd /home/david/cva6/corev_apu/fpga/src/apb_uart/src/slib_counter.vhd /home/david/cva6/corev_apu/fpga/src/apb_uart/src/uart_receiver.vhd /home/david/cva6/corev_apu/fpga/src/apb_uart/src/slib_edge_detect.vhd /home/david/cva6/corev_apu/fpga/src/apb_uart/src/slib_input_sync.vhd /home/david/cva6/corev_apu/fpga/src/apb_uart/src/slib_clock_div.vhd /home/david/cva6/corev_apu/fpga/src/apb_uart/src/slib_fifo.vhd /home/david/cva6/corev_apu/fpga/src/apb_uart/src/uart_baudgen.vhd
# vcom -64 -nologo -quiet -2008 -work work -pedanticerrors 
vlog +cover=bcfst+/dut -incr -64 -nologo -quiet -suppress 13262 -permissive +define+WT_DCACHE+RVFI_TRACE -timescale "1ns / 1ns" -work work -pedanticerrors /home/david/cva6/core/serdiv.sv /home/david/cva6/core/ariane_regfile_ff.sv /home/david/cva6/core/issue_stage.sv /home/david/cva6/core/cvxif_fu.sv /home/david/cva6/core/branch_unit.sv /home/david/cva6/core/dromajo_ram.sv /home/david/cva6/core/controller.sv /home/david/cva6/core/commit_stage.sv /home/david/cva6/core/re_name.sv /home/david/cva6/core/mult.sv /home/david/cva6/core/csr_buffer.sv /home/david/cva6/core/decoder.sv /home/david/cva6/core/ex_stage.sv /home/david/cva6/core/scoreboard.sv /home/david/cva6/core/store_unit.sv /home/david/cva6/core/ariane.sv /home/david/cva6/core/axi_adapter.sv /home/david/cva6/core/fpu_wrap.sv /home/david/cva6/core/csr_regfile.sv /home/david/cva6/core/load_store_unit.sv /home/david/cva6/core/id_stage.sv /home/david/cva6/core/multiplier.sv /home/david/cva6/core/store_buffer.sv /home/david/cva6/core/compressed_decoder.sv /home/david/cva6/core/axi_shim.sv /home/david/cva6/core/alu.sv /home/david/cva6/core/instr_realign.sv /home/david/cva6/core/perf_counters.sv /home/david/cva6/core/cva6.sv /home/david/cva6/core/amo_buffer.sv /home/david/cva6/core/load_unit.sv /home/david/cva6/core/issue_read_operands.sv /home/david/cva6/core/fpu/src/fpnew_fma.sv /home/david/cva6/core/fpu/src/fpnew_opgroup_fmt_slice.sv /home/david/cva6/core/fpu/src/fpnew_divsqrt_multi.sv /home/david/cva6/core/fpu/src/fpnew_fma_multi.sv /home/david/cva6/core/fpu/src/fpnew_opgroup_multifmt_slice.sv /home/david/cva6/core/fpu/src/fpnew_classifier.sv /home/david/cva6/core/fpu/src/fpnew_noncomp.sv /home/david/cva6/core/fpu/src/fpnew_cast_multi.sv /home/david/cva6/core/fpu/src/fpnew_opgroup_block.sv /home/david/cva6/core/fpu/src/fpnew_rounding.sv /home/david/cva6/core/fpu/src/fpnew_top.sv /home/david/cva6/core/fpu/src/fpu_div_sqrt_mvp/hdl/iteration_div_sqrt_mvp.sv /home/david/cva6/core/fpu/src/fpu_div_sqrt_mvp/hdl/nrbd_nrsc_mvp.sv /home/david/cva6/core/fpu/src/fpu_div_sqrt_mvp/hdl/div_sqrt_top_mvp.sv /home/david/cva6/core/fpu/src/fpu_div_sqrt_mvp/hdl/preprocess_mvp.sv /home/david/cva6/core/fpu/src/fpu_div_sqrt_mvp/hdl/control_mvp.sv /home/david/cva6/core/fpu/src/fpu_div_sqrt_mvp/hdl/norm_div_sqrt_mvp.sv /home/david/cva6/core/fpu/src/fpu_div_sqrt_mvp/hdl/div_sqrt_mvp_wrapper.sv /home/david/cva6/core/frontend/frontend.sv /home/david/cva6/core/frontend/instr_scan.sv /home/david/cva6/core/frontend/instr_queue.sv /home/david/cva6/core/frontend/bht.sv /home/david/cva6/core/frontend/btb.sv /home/david/cva6/core/frontend/ras.sv /home/david/cva6/core/cache_subsystem/tag_cmp.sv /home/david/cva6/core/cache_subsystem/wt_dcache_ctrl.sv /home/david/cva6/core/cache_subsystem/amo_alu.sv /home/david/cva6/core/cache_subsystem/wt_axi_adapter.sv /home/david/cva6/core/cache_subsystem/std_nbdcache.sv /home/david/cva6/core/cache_subsystem/cache_ctrl.sv /home/david/cva6/core/cache_subsystem/miss_handler.sv /home/david/cva6/core/cache_subsystem/std_cache_subsystem.sv /home/david/cva6/core/cache_subsystem/wt_dcache_missunit.sv /home/david/cva6/core/cache_subsystem/cva6_icache.sv /home/david/cva6/core/cache_subsystem/wt_dcache_wbuffer.sv /home/david/cva6/core/cache_subsystem/wt_l15_adapter.sv /home/david/cva6/core/cache_subsystem/wt_dcache_mem.sv /home/david/cva6/core/cache_subsystem/wt_cache_subsystem.sv /home/david/cva6/core/cache_subsystem/wt_dcache.sv /home/david/cva6/core/cache_subsystem/cva6_icache_axi_wrapper.sv /home/david/cva6/corev_apu/bootrom/bootrom.sv /home/david/cva6/corev_apu/bootrom/dromajo_bootrom.sv /home/david/cva6/corev_apu/clint/axi_lite_interface.sv /home/david/cva6/corev_apu/clint/clint.sv /home/david/cva6/corev_apu/fpga/src/axi2apb/src/axi2apb_wrap.sv /home/david/cva6/corev_apu/fpga/src/axi2apb/src/axi2apb.sv /home/david/cva6/corev_apu/fpga/src/axi2apb/src/axi2apb_64_32.sv /home/david/cva6/corev_apu/fpga/src/apb_timer/apb_timer.sv /home/david/cva6/corev_apu/fpga/src/apb_timer/timer.sv /home/david/cva6/corev_apu/fpga/src/axi_slice/src/axi_w_buffer.sv /home/david/cva6/corev_apu/fpga/src/axi_slice/src/axi_r_buffer.sv /home/david/cva6/corev_apu/fpga/src/axi_slice/src/axi_slice_wrap.sv /home/david/cva6/corev_apu/fpga/src/axi_slice/src/axi_slice.sv /home/david/cva6/corev_apu/fpga/src/axi_slice/src/axi_single_slice.sv /home/david/cva6/corev_apu/fpga/src/axi_slice/src/axi_ar_buffer.sv /home/david/cva6/corev_apu/fpga/src/axi_slice/src/axi_b_buffer.sv /home/david/cva6/corev_apu/fpga/src/axi_slice/src/axi_aw_buffer.sv /home/david/cva6/corev_apu/src/axi_riscv_atomics/src/axi_riscv_amos.sv /home/david/cva6/corev_apu/src/axi_riscv_atomics/src/axi_ [... cut ...]
vopt +cover=bcfst+/dut -incr -64 -nologo -quiet -suppress 13262 -permissive +define+WT_DCACHE+RVFI_TRACE -work work  ariane_tb -o ariane_tb_optimized +acc -check_synthesis
** Warning: (vopt-13288) Multiple macros defined in +define+ command line switch.
# vopt_stacktrace.vstf written
** Fatal: Unexpected signal: 11.
/home/david/cva6/core/fpu/src/fpu_div_sqrt_mvp/hdl/iteration_div_sqrt_mvp.sv(1): Vopt Compiler exiting
make: *** [Makefile:344: build] Error 232
~~~

I have checked the warnings but the if conditions that check the value of `NR_RGPR_PORTS` seem ok. Any help on what the problem may be would be much appreciated. Thanks!

David

christian-lanius, 2022-03-28:
Dear David,
The same issue has also been raised in #800 I think. Unfortunately, this issue has not been resolved as of yet as far as I am aware. It seems related to the exact release of questasim you are using. I have tried the 2018 and 2020 version, where this issue occurs as well.

davidmallasen, 2022-03-30:
The workaround of #800 works for me as well. However, without optimizations the simulation time is ~5x longer for a simple hello world. Thanks for sharing @christian-lanius ! I can't seem to pinpoint where the issue arises with the new patches (as it was working before commit 00236be3d8552f93a0bebda8f9820ec54b64a000), but if someone manages to find a solution it would be very helpful.

christian-lanius, 2022-04-19:
The issue is related to the AXI interconnect I think. I have replaced the ariane core with a synthesized version and the same issue occurs, so it appears as if the actual core is not the culprit (at least not the coding style).
I have looked at the axi repo and there I can run the test-benches for the interconnect just fine, so it must break somewhere in between, maybe the exact configuration of the interconnect. But I was unable to look too deeply into it.

niwis, 2022-11-29:
This seems to be a vopt bug in newer versions of questasim. They struggle with parametric data types, which are used in several of our IPs. I get the same segfault for questasim version `2021.1`, but not for `2020.3`.

christian-lanius, 2022-11-30:
I can confirm that this is questasim version related: Versions 10.7c generates a different error message, versions 2020.4 and 2021.3 generate the discussed segmentation fault.
According to @niwis version 2020.3 works and I can confirm that version 2020.4 (which is the closest I have available) works as well.
I suppose this can only be fixed moving forward by updating the readme with some info about this and the makefile with some error/warning message after checking the questasim version. Additionally, a bug report ought to be filled with Siemens/Mentor about this.

christian-lanius, 2023-01-23:
So Europractice has made available a new version of Questasim (2022.4) and that version does indeed also fixes this issue! 
I mainly add this comment here for people trying to figure out which versions work. So I hope that this error does not reappear in the future with future versions of Questasim.

MikeOpenHWGroup, 2023-02-11:
Hi @christian-lanius and @davidmallasen.  It appears that this issue has been resolved.  Please confirm, and if so close the issue, otherwise this issue will be closed by 2023-02-24.  Thanks!

davidmallasen, 2023-02-13:
I haven't been able to get my hands on the newer Questasim version, so I can't test it yet. However, if @christian-lanius tested version 2022.4 and it solves the issue, we can close it for now.

closed 2023-02-13 by davidmallasen