openhwfoundation/cva6 #844: Debug mode exception

dpetrisko, 2022-03-26:
Hi, I had a question on the functionality of debug mode in the core. Let me know if this is a more appropriate question for https://github.com/pulp-platform/riscv-dbg. When cva6 is in debug, running a program in the program buffer and encounters an exception, how does it communicate to the debug module to set cmderr?

[image]

I was able to trace it back to here in the debug module:
https://github.com/pulp-platform/riscv-dbg/blob/d83a5f9547bd04d4fce47b4c91cc03e502ab5607/src/dm_mem.sv#L257
https://github.com/pulp-platform/riscv-dbg/blob/d83a5f9547bd04d4fce47b4c91cc03e502ab5607/src/dm_mem.sv#L48

which corresponds to here:
https://github.com/openhwgroup/cva6/blob/c26bda3f7bd2291a8704f27a1088b9587e9b4c4b/corev_apu/tb/ariane_testharness.sv#L238

But then I was unable to trace that back past the AXI bus.  Is there some logic in cva6 which is sending this debug req message when an exception occurs, or is the debug module somehow detecting by virtue of the sequence not successfully completing?

niwis, 2022-03-28:
Hi @dpetrisko, in debug mode, the trap vector is set to the debug ROM:
https://github.com/openhwgroup/cva6/blob/c26bda3f7bd2291a8704f27a1088b9587e9b4c4b/core/csr_regfile.sv#L1002-L1005
There, the `EXCEPTION` word of the `dm_mem` is written to, which raises the `cmderr` flag (in the lines of `dm_mem` that you posted):
https://github.com/pulp-platform/riscv-dbg/blob/d83a5f9547bd04d4fce47b4c91cc03e502ab5607/debug_rom/debug_rom.S#L86

niwis, 2022-03-28:
Yes, `dm::ExceptionAddress` is a constant defined in the debug module package:
https://github.com/pulp-platform/riscv-dbg/blob/d83a5f9547bd04d4fce47b4c91cc03e502ab5607/src/dm_pkg.sv#L30
`0x800` is the debug ROM base, and `8` is the offset of the exception entry point:
https://github.com/pulp-platform/riscv-dbg/blob/d83a5f9547bd04d4fce47b4c91cc03e502ab5607/debug_rom/debug_rom.S#L31-L32

Changes to the debug ROM only require an update to the `dm_pkg` if the entry points change (which shouldn't be the case unless you explicitly do so).

As far as I know, these addresses are implementation-specific and not standardised. In general, the debug module follows the execution-based approach described in A.2 of the RISC-V debug spec.

dpetrisko, 2022-03-29:
Thanks for the clarification!

closed 2022-03-29 by dpetrisko