openhwfoundation/cva6 #845: Support for MCYCLEH, MINSTRETH, CYCLEH, TIMEH and INSTRETH in 32-bit mode

spersvold, 2022-03-29:
Hi,

In 32-bit mode it would be nice to get access to the full 64-bit wide cycle and instruction counters for precision timing. Currently there's an exception of code references these CSR registers.

I'm attaching a patch that adds support for this. Please consider.
[0001-Added-support-for-MCYCLEH-MINSTRETH-CYCLEH-TIMEH-and.patch.txt](https://github.com/openhwgroup/cva6/files/8375627/0001-Added-support-for-MCYCLEH-MINSTRETH-CYCLEH-TIMEH-and.patch.txt)

Thanks!
-- Steffen

spersvold, 2022-03-30:
I've also added a pull request #847

MikeOpenHWGroup, 2023-02-17:
A pull-request to address this issue (#847) has been merged in, so I will close this issue.

closed 2023-02-17 by MikeOpenHWGroup