openhwfoundation/cva6 #856: read/access the address of specific CSR register

sreersr, 2022-04-11:
How to read/access the address of specific CSR register listed in https://github.com/openhwgroup/cva6/blob/master/docs/user_guide/cva6_ug_csr.adoc
In other words how to visualize these CSR registers
Please help me in this regard

MikeOpenHWGroup, 2023-02-17:
Hi @SreeRanjaniRajendran, I see that nobody has responded to this issue in ~10 months.  In that time, the CSR documentation has been significantly improved and more improvement are on-going.  Please have a look in the `CV32A6_CSR programmers view` of the [CVA6 User Manual](https://docs.openhwgroup.org/projects/cva6-user-manual/index.html).

If this meets your needs, please close this Issue.  If not, let us know what you need.

If you do not close this Issue by 2023-03-02, we will close it.

MikeOpenHWGroup, 2023-03-13:
Closing due to a lack of activity.

closed 2023-03-13 by MikeOpenHWGroup