openhwfoundation/cva6 #858: cv32a6 benchmarks - Cross-compiler options

pcotret, 2022-04-12:
I was playing with a `cv32a6` variant.
In `riscv-tests`, while ISA tests are compiled for both 32-bit and 64-bit, benchmarks will only be compiled for 64-bits when running https://github.com/openhwgroup/cva6/blob/master/ci/build-riscv-tests.sh

I'm trying to find an elegant to do it without breaking everything: it should be a modification in `riscv-tests` Makefile.

MikeOpenHWGroup, 2023-02-17:
Hi @pcotret.  I see that nobody has responded to this issue in ~10 months.  Is this still a question for you?  If so, please update this issue and I will assign it to the appropriate person.  Otherwise please close it.

If we do not hear from you by 2023-03-02 we will close the issue.

pcotret, 2023-02-20:
Didn't rework on this since April 22... Issue closed.

closed 2023-02-20 by pcotret