openhwfoundation/cva6 #859: Verify the core on Questa

FranDuqueAyachi, 2022-04-17:
Hi!
Im a completly beginner on this world and my knowledge about it its very poor because I'm still studying the degree so sry if my questions are "simple".
From now the only thing that I want to do its to verify the cva6 core, Im following the next steps to do it:
**1- obviously first thing its to get the core:**
git clone https://github.com/openhwgroup/cva6.git
git submodule update --init --recursive
**2- get the tools so Im able to compile the core:**
$ git clone https://github.com/riscv/riscv-tools
$ cd riscv-tools
$ git submodule update --init --recursive
$ export RISCV=
$ export PATH=${PATH}:${RISCV}/bin
$ ./build.sh
**3- Have all neccesary to work (my case RHEL)**
$ sudo yum install autoconf automake git libmpc-devel mpfr-devel gmp-devel gawk bison flex texinfo gcc gcc-c++ zlib-devel

So my questions are:

1. to verify the module with questasim what should I follow in terms of help to do it, because all I see its verilator and UVM but what I need its to use the test that are on the repository and simulate them with questasim.
2. Is necessary to build all the tools as I did to just verify the core?, because when I make the comand $ ./build.sh it generate an error that says that my version of automake isnt 1.14 or superior but actually I got the version 1.14 of automake so how can I fix this error or if I can find another way to just build the tools that I need to verify the module.
Sry again if my question could be "stupid" I have just to verify the module and supposedly I have everything I need to do this job.
Thanks a lot, anything could help!

davidmallasen, 2022-04-17:
Have a look at Issue #842 if you want to simulate with QuestaSim. There are currently some problems with that workflow and the simulations end up being quite slow.

FranDuqueAyachi, 2022-04-17:
> Have a look at Issue #842 if you want to simulate with QuestaSim. There are currently some problems with that workflow and the simulations end up being quite slow.

Thanks for your answer, so then, if I want to simulate the core and verify the core on QuestaSim isnt a good Idea?
In this case, how can I just simulate the core using differents test, I just want to verify some tipycal funcionalities from a RISC-V core such simple as operations and things like that.
Maybe with verilator it works? 
Thanks again

davidmallasen, 2022-04-18:
You can do it with QuestaSim, but until there is a fix it will run slow. I don't know about the flow running the RISC-V tests, although if you have your own tests as a .elf you can follow the instructions in the readme or the issue I mentioned. With verilator it should work too as far as I know.

christian-lanius, 2022-04-19:
@FranDuqueAyachi You can "verify" the core in Questasim if you pay attention to the discussions in #800 and #842. There is a make targets for the different parts. If you run the command "make -j8 run-asm-tests run-amo-tests run-mul-tests run-fp-tests run-benchmarks" (modify the number of processes to the number of licenses you have available), the different tests will be executed. You have to create a "tmp" folder beforehand I think.

closed 2022-05-03 by FranDuqueAyachi

christian-lanius, 2022-08-02:
Is the issue with vopt crashing going to be fixed or is this work around considered "good enough"? It seems to me like it is not feasible to work with questasim with the core at this time because of the slow simulation speed.