openhwfoundation/cva6 #877: No makefile in core/example_tb

joeya20, 2022-05-16:
The readme instructs to 
```
$ cd core/example_tb
$ make veri_run
```
to run the core example testbench but there is no makefile in core/example_tb.

pjlobo, 2023-01-16:
Just found this in commit https://github.com/openhwgroup/cva6/commit/3ddf797e95923fd11113c8e443046105dfbf8843 dated Sep 24, 2021:

>* Move example_tb from cva6 to core-v-verif project
>
>Signed-off-by: Jean-Roch Coulon <jean-roch.coulon@thalesgroup.com>

And it seems to correlate with https://github.com/openhwgroup/core-v-verif/commit/56f43b03d243dcb2b1ebd050dd6cdbe11ed6ee78 dated Jun 21, 2021:

> install-riscv-dv.sh: move example_tb (from cva6 project) to cva6/core/tb
> Signed-off-by: Jean-Roch Coulon <jean-roch.coulon@thalesgroup.com>

Apparently this movement was part of a bigger repository reorganization and it was not reflected in the README.

pjlobo, 2023-01-16:
Just found this in commit https://github.com/openhwgroup/cva6/commit/3ddf797e95923fd11113c8e443046105dfbf8843 dated Sep 24, 2021:

>[...]
>* Move example_tb from cva6 to core-v-verif project
>
>Signed-off-by: Jean-Roch Coulon <jean-roch.coulon@thalesgroup.com>
>[...]

And it seems to correlate with https://github.com/openhwgroup/core-v-verif/commit/56f43b03d243dcb2b1ebd050dd6cdbe11ed6ee78, dated Jun 21, 2021:

>install-riscv-dv.sh: move example_tb (from cva6 project) to cva6/core/tb
>Signed-off-by: Jean-Roch Coulon <jean-roch.coulon@thalesgroup.com>

It seems that the move was part of a bigger repository reorganization and among so many changes the README update was forgotten.

pjlobo, 2023-01-16:
Support for running the example testbench with Verilator was removed a few days later in https://github.com/openhwgroup/core-v-verif/commit/230b19042d64b266cb216c9d495e138113d3eb16, so farewell. I've been unable to find any mention to the rationale. Perhaps lack of UVM support in Verilator, but that's just a guess.

The stale README problem, however, continues to live in core-v-verif: The README in https://github.com/openhwgroup/core-v-verif/tree/master/cva6 still mentions a Verilator-related environment variable.

MikeOpenHWGroup, 2023-02-11:
Hi @pjlobo, thanks for the issue and my apologies for not responding for so long.  The OpenHW verification team does not support the testbench in this repository, so please let me know if you are still having trouble with that and I will find the right person to help you.

The UVM environment for CVA6 under [CORE-V-VERIF](https://github.com/openhwgroup/core-v-verif) is stable.  You are right that the UVM testbench does not run with Verilator -- you will need a commerical simulator for that.

Let me know if your issues are resolved by either closing this issue or providing an update.  I promise not to leave the issue un-attended for so long.

MikeOpenHWGroup, 2023-03-06:
There has been not update since my last comment three weeks ago so I am closing this issue.  Please feel free to re-open it or (better yet) create a new one.

closed 2023-03-06 by MikeOpenHWGroup