openhwfoundation/cva6 #879: [Bug Report] Issue with memory access

chenc6, 2022-05-22:
Hi,
We found that cva6 allows LOAD instructions to load value from an address that is outside the DRAM space. According to RISC-V Unprivileged ISA V20190608, page 7, "if an instruction attempts to access memory at an inaccessible address, an exception is raised for the instruction. Vacant locations in the address space are never accessible". However, cva6 did not throw an exception for this case.
We have the triggering test case (.mem file), the RTL trace log, and the spike trace log attached below. We can see that we are trying to load from an address outside DRAM space to register x10 (lbu  a0, -302(sp)). CVA6 is not throwing any exception for this instruction whereas spike is throwing load access fault exception.

Is this a bug in cva6?
[bug_report.zip](https://github.com/openhwgroup/cva6/files/8749872/bug_report.zip)

zarubaf, 2022-05-23:
I wouldn't say a bug, but rather an unsupported feature. Initially, there was some push to get this feature in but I think that never came to fruition. Adding @JeanRochCoulon to see whether this is something that could be interesting to add (I would think this is a very useful feature).

MikeOpenHWGroup, 2023-02-11:
Hello @JeanRochCoulon, it seems the ball is in your court for this issue.  If you decide to _not_ to support this feature, then please close this issue and create a task to clarify the behaviour in the User Manual.   Otherwise, and maybe this is better, you could decide to support this feature and so change the label from `Type:Enhancement` to `Type:Bug`.

JeanRochCoulon, 2023-03-16:
Indeed @MikeOpenHWGroup It is interesting to support this feature. WANTED VOLUNTEER !! We can keep this Issue as Type:Enhancement.

JeanRochCoulon, 2025-01-03:
@AyoubJalali As you know, we are cleaning the "Github Issues". Have you verified this issue doing trap verification ?   What is the status ?

AyoubJalali, 2025-01-03:
> @AyoubJalali As you know, we are cleaning the "Github Issues". Have you verified this issue doing trap verification ? What is the status ?

for now all exceptions are covered except : 
- INSTRUCTION ACCESS FAULT
- LOAD ACCESS FAULT
- STORE ACCESS FAULT

The reason why ? : PMP was attached to MMU (CV32A65X doesn't support MMU), but for now we extract PMP from MMU and we can cover these exceptions when we start verifying PMP checks.

JeanRochCoulon, 2025-01-03:
As Today PMP is available, it would be helpful to add PMP-TRAPS tests to cover the 3 exception types. To be discussed with @ASintzoff. Good for code coverage too.

AyoubJalali, 2025-01-03:
> As Today PMP is available, it would be helpful to add PMP-TRAPS tests to cover the 3 exception types. To be discussed with @ASintzoff. Good for code coverage too.

Indeed and I tried doing that, I failed because I think there's a Bug :) to be investigate !

AyoubJalali, 2025-01-16:
we should see if Spike has a option to give access to all memory address space, because some tests failed due to this mismatch @zchamski