openhwfoundation/cva6 #880: [Bug Report] CPU mode is not provided to the CVXIF interface

ZElkacimi, 2022-05-23:
When executing a simple test offloading a cvxif instruction (example: with core-v-verif, use the following command to execute cvxif_multi.S test; `python3 cva6.py --testlist=../tests/testlist_cvxif.yaml --test cvxif_multi  --iss_yaml cva6.yaml --target cv64a6_imafdc_sv39 --iss=vcs-uvm`), I notice at the log file that cva6 mode is 3 (machine mode).
Following is a snippet of the csv output file:
[image]

While in the message printed by the agent’s monitor, the mode offloaded by the cvxif interface (issue_req.mode) is always 0.
In addition, Looking at the rtl code (cvxif_fu module), mode signal is never assigned.

MikeOpenHWGroup, 2023-02-11:
Hi @ZElkacimi, this issue is more than 7 months old and has not received any updates. It looks like an important issue to me. Please work with @JeanRochCoulon to assign the right person to move this along.

AyoubJalali, 2023-07-20:
This issue has been fixed by PR #1298

closed 2023-07-20 by AyoubJalali