openhwfoundation/cva6 #882: Fail run on Questasim libfesvr.so cannot open sahred object

FranDuqueAyachi, 2022-05-24:
Hi!
I'm trying to simulate the hello.elf on questasim with the command:
**make sim elf-bin=$RISCV/riscv64-unknown-elf/bin/pk target-options=hello.elf  batch-mode=1**
My Questasim version 2019.2 Linux_x86_64
It seems to run and start loading all modules "work."
then it generate the UVM-1.1d enviroment generating the file trace_hart_o.log with all the instructions and after 40000000 ns it ends but with 2 errors wich are:
Loading /home/users/PFC/divsicad/fduque/tfg/riscv/lib/libfesvr.so
# ** Error (suppressible): (vsim-3197) Load of "/home/users/PFC/divsicad/fduque/tfg/riscv/lib/libfesvr.so" failed: /home/users/PFC/divsicad/fduque/tfg/riscv/lib/libfesvr.so: cannot open shared object file: No such file or directory.
when installed libfesvr by ./ci/install-libfesvr it didnt install any file .so
How can I fix this error, thank you so much!

fpetrot, 2022-12-20:
Hello,
better later than never: have a look at https://github.com/openhwgroup/cva6/issues/800
Frédéric

MikeOpenHWGroup, 2023-02-11:
Hi @fpetrot, this issue has not received a comment for almost two months.  Is it still a problem for you?  Please comment and if the problem is resolved, close the issue.  Otherwise, this issue will be closed by 2023-02-24.   Thanks!

MikeOpenHWGroup, 2023-02-27:
We have not heard back from @fpetrot, so I am closing this issue.

closed 2023-02-27 by MikeOpenHWGroup