openhwfoundation/cva6 #883: [Bug Report] Wrong type exception when access invalid memory

chenc6, 2022-05-24:
Hi,

Our test cases show that when cva6 jumps to an invalid address (e.g., 0x7ff84538), it throws an Instruction Page Fault exception (at 22990ns). However, we believe it should throw the Instruction Access Fault exception based on the priority of exception from the  RISC-V ISA Volume II: Privileged Architecture 20211203.

[image]

[image]

The test mem file and RTL log are attached.
[wrong_excep.zip](https://github.com/openhwgroup/cva6/files/8766313/wrong_excep.zip)

MikeOpenHWGroup, 2023-02-11:
Hi @JeanRochCoulon, this issue is more than 7 months old and has not received any updates. It looks like an important issue to me. Please assign the right person to move this along.

Jbalkind, 2023-02-18:
It's possible that this was addressed by the MMU changes in #968 but I'm not 100% sure. Worth checking

JeanRochCoulon, 2023-03-15:
Hi André, as this GitHub issue is related to the traps, I feel good to assign it to you.

ASintzoff, 2023-11-14:
According to Table 3.7 of RISC-V ISA Volume II: Privileged Architecture 20211203, the priority of page fault or access fault during instruction address translation is determined by the first one encountered.
Therefore the CVA6 behaviour is compliant with RISC-V specification.

BTW the exception cause code is not related at all to the exception priority.

closed 2023-11-14 by ASintzoff