openhwfoundation/cva6 #884: [Bug Report] Out of bound array access

chenc6, 2022-05-24:
Hi,

CVA6 RTL is returning unknown value ("xxxxxxxx") when reading the value of MHPMCOUNTER30. This is happening because the csr reg data (```csr_rdata```) is being read from ``perf_counter_d`` array in the perf_counters.sv module with an array index outside its range. The range of ``perf_counter_d`` is [riscv::CSR_MIF_EMPTY : riscv::CSR_ML1_ICACHE_MISS] (line:46 from perf_counter.sv file),  but in line: 282 of csv_regfile.sv we are using the array to access array index  riscv::CSR_MHPM_COUNTER_30, which is greater than the range of [riscv::CSR_MIF_EMPTY : riscv::CSR_ML1_ICACHE_MISS]. 

Please find attached the triggering input mem file, spike_trace_log, and cva6_trace_log. We can see from the line:83 of cva6 trace log that the read data is ``xxxxxxxx``
[image]

[outofrange.zip](https://github.com/openhwgroup/cva6/files/8766647/outofrange.zip)

MikeOpenHWGroup, 2023-02-11:
Hi @JeanRochCoulon and @yanicasa , this issue is more than 7 months old and has not received any updates. It looks like an important issue to me. It could be that this issue has been already resolved with the latest updates to the Tracer.

JeanRochCoulon, 2023-03-15:
Hello @TulikaSi   This GitHub issue is related to perf counters. As you are modifying the perf counters, maybe the bug will be naturally removed. If it is not the case, can you fix it.

JeanRochCoulon, 2023-03-15:
@MikeOpenHWGroup   I (re)assigned you this GitHub Issue because Tulika does not appear in the assignee list. Do we need to add her to the project list ?

TulikaSi, 2023-03-16:
> Hello @TulikaSi This GitHub issue is related to perf counters. As you are modifying the perf counters, maybe the bug will be naturally removed. If it is not the case, can you fix it.

Hi @JeanRochCoulon the hardware performance monitor includes 29 number of  64-bit event counters, mhpmcounter3–
mhpmcounter31 but in accordance to the requirement specification [https://github.com/openhwgroup/cva6/blob/master/docs/02_cva6_requirements/cva6_requirements_specification.rst#performance-counters](url) the implementation shall take into account only six generic 64-bit performance counters  **mhpmcounter3 to mhpmcounter8**  including their upper 32 bits counterparts **mhpmcounter3h to mhpmcounter8h**. So I believe the the RTL after modification of the performance counter will still be returning unknown value ("xxxxxxxx") when reading the value of MHPMCOUNTER30(as it won't be implemented)

TulikaSi, 2023-12-11:
Hi @JeanRochCoulon 
Since the performance counters have been modified this issue now no longer exists as all the mhpmcounters have been implemented.
We can close this issue.

JeanRochCoulon, 2023-12-11:
@chenc6  If you are agree with @TulikaSi status, can you close the issue ?

chenc6, 2023-12-13:
Sure, thank you for letting me know.

closed 2023-12-13 by chenc6