openhwfoundation/cva6 #885: [Bug report] miss illegal instruction exception when rd of MULHU is the same as rs1 or rs2

chenc6, 2022-05-25:
Hi,

The RISC-V ISA Volume I, 20190608, page 43 mentions: "MULH[[S]U] rdh, rs1, rs2; MUL
rdl, rs1, rs2 (source register specifiers must be in same order and rdh cannot be the same as rs1 or rs2)"
[image]

Our testcase shows cva6 will not throw illegal instruction exception in this case while spike does at line 84.
[image]

The testcase mem file, rtl trace log, and spike trace log are attached.
[missexcept.zip](https://github.com/openhwgroup/cva6/files/8767083/missexcept.zip)

Thank you

MikeOpenHWGroup, 2023-02-11:
Hi @JeanRochCoulon , this issue is more than 7 months old and has not received any updates. It _looks_ like an important issue to me.  Please assign the right person to move this along.

JeanRochCoulon, 2023-03-14:
@AyoubJalali  You should meet this case doing the ISACOV verification. Does ISACOV cover this case ?

AyoubJalali, 2023-03-14:
@JeanRochCoulon It should be, but i'll look at it

AyoubJalali, 2023-03-14:
@chenc6 hello, so I run a test with : 
_**main:             mulhsu     zero, zero, zero
                  mul        zero, zero, zero
                  slti       zero, t6, -972
                  div        ra, s2, s8
                  andi       s0, s9, -182
                  or         t6, a3, a5
                  auipc      t4, 691756
                  mul        tp, s3, t1
                  add        t0, a6, zero
                  srli       a7, a2, 7
                  sub        s5, s11, s11
                  srai       s7, s4, 26
                  or         a2, sp, t1
                  remu       ra, t2, a1
                  divu       a3, sp, s1
                  slli       s3, s4, 2
                  srl        s4, s3, t2
                  rem        tp, s1, a3
                  div        ra, t5, t5
                  j test_done**_
and i didn't get any illegal instructions exception on spike side as you see here : 

[image]

so could you providing me with the source file .S, to try to reproduce the problem if it still.
@JeanRochCoulon just to answering your question it's cover as far as I know

chenc6, 2023-03-14:
@AyoubJalali Sorry, I couldn't find the .S file, but it should be easy to reproduce because the issue is only related to one instruction.

Maybe you can change the instruction from `mulhsu` to `mulhu`, such that the binary code will be 0200`3`033 but not 0200`2`033?

AyoubJalali, 2023-03-14:
@chenc6 I don't thing also it's a problem, because i covered all possible combinations between all multiplications instructions with all values of rx registers :

[image]

So could the problem be in spike version, so which version you use for this test ?

JeanRochCoulon, 2023-03-15:
@chenc6 and @AyoubJalali   The RISCV specification tells us that "rdh cannot be the same as rs1 or rs2".    It is not clear to me if an illegal instruction must be triggered or not. Does the RISCV or the CVA6 ISA specification tells us what is the behaviour in that case?

ASintzoff, 2023-03-15:
@JeanRochCoulon @chenc6 @AyoubJalali The CVA6 behaves correctly. There is no reason to raise an illegal instruction exception on `MULH[[S]U]` instruction. 

The specification only warns the software developer when using `MULH[[S]U]` and `MUL` instructions. If `rdh` is the same as `rs1` or `rs2` for `MULH[[S]U]`, then the following `MUL` will compute a result based on a modified `rs1` or `rs2`. Therefore, in such case, the `MUL` result will not be the expected one.

AyoubJalali, 2023-03-15:
@JeanRochCoulon I don't think the cva6 should raised an illegal instruction it's just a recommendation if you want the lower & upper part of the multiplication you should use a sequence mentioned in the RISCV spec.

JeanRochCoulon, 2023-03-15:
Understood. Thank for your investigation.

AyoubJalali, 2023-11-02:
After the discussion above, I think there's no issue related to the instructions mentioned above

closed 2023-11-02 by AyoubJalali