openhwfoundation/cva6 #886: torture test failed due to sbt version and +signature option

timingwalker, 2022-05-25:
I try to run CI test of  travis-ci-emul.sh in my local environment. When i run torture-rtest-verilator, there are two problem:

1. make torture-gen
touture  version in get-torture.sh is 59b0f0  which using sbt0.13. It is too old and makes torture-gen fail.
the latest version in torture repository bump sbt to 1.4.4 and fix this problem: https://github.com/ucb-bar/riscv-torture/pull/29

2. make run-torture-verilator
the report shows that load_elf failed: 
> work-ver/Variane_testharness +max-cycles=10000000 +signature=tmp/riscv-torture/output/test.rtlsim.sig tmp/riscv-torture/output/test
> This emulator compiled with JTAG Remote Bitbang client. To enable, use +jtag_rbb_enable=1.
> Listening on port 45859
> Variane_testharness: ../fesvr/elfloader.cc:22: std::map<std::__cxx11::basic_string<char>, long unsigned int> load_elf(const char*, memif_t*, reg_t*): Assertion `fd != -1' failed.
> make: *** [Makefile:758: run-torture-verilator] Error 255

I think it is caused by the +signature option in makefile, whick is not an option in cva6/corev_apu/tb ariane_tb.cpp.
https://github.com/openhwgroup/cva6/blob/75807530f26ba9a0ca501e9d3a6575ec375ed7ab/Makefile#L758

Delete +signature can fix this problem, but check-torture fails because rtlsim.sig is not generated.

So is torture test still using in git hub CI test?

MikeOpenHWGroup, 2023-02-11:
Hi @yanicasa, can you comment on this issue?

JeanRochCoulon, 2025-01-02:
torture suite is not supported by CV32A65X, add notCV32A65X label