openhwfoundation/cva6 #887: CVXIF feature missing: wait for the result of the instruction that raise the issue resp.exc signal

ZElkacimi, 2022-05-25:
CPU shall wait for the result of an instruction that raises the issue_resp.exc=1 signal (by the coprocessor), because it can then no longer retire (or commit via the commit interface) any younger instruction before it is known if the possible exception in the coprocessor actually happened or not, which will only be known at the corresponding result handshake (result.exc).
That is the easiest implementation; otherwise CPU might need to unroll/undo instructions that should not have been executed.

MikeOpenHWGroup, 2023-02-11:
Hi @ZElkacimi, this issue is more than 6 months old and has not received any updates. Please work with @JeanRochCoulon to find the right person to move this along.

Gchauvon, 2023-12-06:
Still relevant

AyoubJalali, 2024-12-24:
The protocol has been updated to higher version v1.0.0
This issue is related to last version of protocol

closed 2024-12-24 by AyoubJalali