openhwfoundation/cva6 #891: CVA6 never raise the cvxif commit_kill signal

ZElkacimi, 2022-05-27:
If commit_valid is 1 and commit_kill is 0, then the core guarantees that the offloaded instruction (id) is no longer speculative, will not get killed (e.g. due to misspeculation or an exception in a preceding instruction), and is allowed to be committed. If commit_valid is 1 and commit_kill is 1, then the offloaded instruction (id) shall be killed in the coprocessor and the coprocessor must guarantee that the related instruction does/did not change architectural state.

In all tested cvxif custom instructions, cva6 have never raised the commit_kill signal, even when it has an exception in a preceding instruction.

MikeOpenHWGroup, 2023-02-11:
Hi @ZElkacimi, this issue is more than 6 months old and has not received any updates. Pleas work with @JeanRochCoulon to find the right person to move this along.

Gchauvon, 2023-12-06:
Feature not supported. Not a bug.

closed 2023-12-06 by Gchauvon