openhwfoundation/cva6 #895: CVA6 do not support cvxif dualwrite/dualread

ZElkacimi, 2022-06-02:
CVA6 do not support dualread and dualwrite from a Coprocessor.

Also, related to dualwrite: It is mentioned in the cvxif spec, that **we** is 2 bits wide when XLEN = 32 and X_RFW_WIDTH = 64, and 1 bit wide otherwise. If **we** is 2 bits wide, then we[1] is only allowed to be 1 if we[0] is 1 as well (i.e. for dual writeback). The picture below, from cvxif spec, shows how it should be declared:

[image]

While in cva6, at cvxif_pkg.sv, the **we** signal is declared as follow: 
[image]

We can notice two issues here, the fact that **we** is not declared correctly, and that, definitely, cva6 currently do not support dualwrite from a coprocessor.

MikeOpenHWGroup, 2023-02-11:
Hi @ZElkacimi, this issue is more than 6 months old and has not received any updates. Pleas work with @JeanRochCoulon to find the right person to move this along.

Gchauvon, 2023-12-06:
Feature not supported. Package should be updated to respect the specification more closely.
Specification is in development so a fix might come with an updated implementation of CVXIF

closed 2023-12-06 by Gchauvon