openhwfoundation/cva6 #906: [Bug Report] Incorrect exception type of PMP violation during address translation

Phantom1003, 2022-06-08:
Our co-simulation framework found that the exception type of address translation PMP violation is incorrect.

In the following test case, we let a non-leaf (level 2) PTE out of the PMP range.
Next, we will perform a storage operation where the accessed virtual address will try to access a PTE that does not match any PMP during the translation.
cva6 throws a load access fault, while spike throws a store access fault.

```
[cva6]  Exception @    119000, PC: 000000008000036c, Cause: Load Access Fault,
[cva6]                                 tval: 0000000080102008
[spike] core   0: 0x000000008000036c (0x00b52023) sw      a1, 0(a0)
[spike] core   0: exception trap_store_access_fault, epc 0x000000008000036c
[spike] core   0:           tval 0x0000000040201010
[spike] core   0: 0x0000000080000004 (0x34302f73) csrr    t5, mtval
[error] WDATA SIM 0000000040201010, DUT 0000000080102008
[error] check board clear 30 error
```

According to riscv-privileged specification:
> If accessing pte violates a PMA or PMP check, raise an access-fault exception corresponding to the original access type.

Hence, cva6 should throw a store access fault.

[cva6-9.zip](https://github.com/openhwgroup/cva6/files/8862145/cva6-9.zip)

> @ProjectDimlight helps reproduce the problem

Moschn, 2022-06-09:
Thanks for reporting this. 

I proposed a fix in #908. Can you try if this solves your issues?

zarubaf, 2022-06-09:
Thanks @Moschn. 
@Phantom1003 It would be really great if we could test the patch of @Moschn.

Phantom1003, 2022-06-09:
cva6 throws a correct Store Access Fault exception, but the *tval seems filled with the physical address instead of the virtual address:
```
Exception @    119000, PC: 000000008000036c, Cause: Store Access Fault,
                                tval: 0000000080102008
core   0: 0x000000008000036c (0x00b52023) sw      a1, 0(a0)
core   0: exception trap_store_access_fault, epc 0x000000008000036c
core   0:           tval 0x0000000040201010
core   0: 0x0000000080000004 (0x34302f73) csrr    t5, mtval
[error] WDATA SIM 0000000040201010, DUT 0000000080102008
[error] check board clear 30 error
[CJ]           0 integer register Judge Failed
    1204ns     1189 M 0000000080000004 0 34302f73 csrr           t5, mtval
```

The related verse in specification is:
> When page-based virtual memory is enabled, mtval is written with the faulting virtual address, even for physical-memory access-fault exceptions.

Phantom1003, 2022-06-09:
Yes, I agree with your understanding. 
And the test case passed, cva6 and spike have the same behavior.
Thank you for your quick response, I will close this issue after the commit has been merged.

PR #908 (2022-07-08): Fix exception type on PMP check during PTW
Fixes #906

According to the spec:
> If accessing pte violates a PMA or PMP check, raise an access-fault
> exception corresponding to the original access type.

Found by @Phantom1003 and @ProjectDimlight
commit 011edf49 (2022-07-08): Fix exception type on PMP check during PTW (#908)

Fixes #906

According to the spec:
> If accessing pte violates a PMA or PMP check, raise an access-fault
> exception corresponding to the original access type.

Found by @Phantom1003 and @ProjectDimlight
closed 2022-06-09 by Phantom1003