openhwfoundation/cva6 #910: [TASK] Integration with latest Kernel from FreeRTOS.org

anjaliigedam, 2022-06-10:
### Is there an existing CVA6-SDK task for this?

- [X] I have searched the existing task issues

### Task Description

There is separate RISCV Portable area which has architectural level details

### Description of Done

Testcases working on CVA6 Genesys-2

jquevremont, 2022-09-16:
Outstanding: Ethernet driver; resource to identify.

MikeOpenHWGroup, 2023-02-07:
@jquevremont, @anjaliigedam, what is the status of this issue?

jquevremont, 2023-02-08:
On hold for maternity leave.

jquevremont, 2023-04-18:
Not yet resumed task.

anjaliigedam, 2023-06-20:
"older" FreeRTOS working on recent CVA6.

Have migrated to use latest Kernel from FreeRTOS.org and get demo working. 
Execution stuck at initialization.

anjaliigedam, 2023-10-10:
cva6 FreeRTOS is up-streamed on AWS FreeRTOS repository.
[https://github.com/FreeRTOS/FreeRTOS-Partner-Supported-Demos/tree/44b2426a9bba9456a8f7d0dfe2a5f849122f18be/RISC-V_cva6](url)

We can mark this task as closed.

closed 2023-10-10 by anjaliigedam