openhwfoundation/cva6 #911: [TASK] Feature to transform cache ways into a scratchpad

TulikaSi, 2022-06-10:
### Is there an existing CVA6-SDK task for this?

- [X] I have searched the existing task issues

### Task Description

Demonstration of mechanism to transform cache ways into a scratchpad

### Description of Done

>>A successful demo
>> Zero impact of change on performance
>>A Pull request

jquevremont, 2022-09-16:
Ongoing investigations.

jquevremont, 2022-10-28:
Decided to separate scratchpad from caches for efficiency. RTL coded. Debugging in progress.
Specification updated accordingly: https://github.com/openhwgroup/cva6/blob/master/docs/specifications/cva6_requirement_specification.rst

jquevremont, 2023-01-20:
Write accesses OK. Read accesses currently fail. Features to be presented on Feb. 4th meeting.

MikeOpenHWGroup, 2023-02-07:
> Features to be presented on Feb. 4th meeting.

@jquevremont @TulikaSi, what is the status of this issue?

TulikaSi, 2023-02-08:
Hi @MikeOpenHWGroup There will be a presentation on Scratchpad this week i.e on 10/02/2023

MikeOpenHWGroup, 2023-02-27:
@jquevremont @TulikaSi, what is the status of this issue?  Was it presented, and did we reach a conclusion?

jquevremont, 2023-04-18:
An issue solved. An issue came up.

TulikaSi, 2023-06-20:
Analysed the bug and it is mostly happening due to some issue while integrating with the Frontend module.
Working on modifying the integration with the Frontend module.

jquevremont, 2023-11-21:
We are waiting for Bosch's PR on the scratchpad.