openhwfoundation/cva6 #915: [TASK] Integrate Linux Boot on FPGA to Thales-CI Verification

Gchauvon, 2022-06-10:
### Is there an existing CVA6-SDK task for this?

- [X] I have searched the existing task issues

### Task Description

Integrate Linux Boot on Genesys2 FPGA to Thales-CI Verification.

- Commit necessary scripts for the task
- Result are visible on dashboard

### Description of Done

Every Pull Request on CVA6 trigger a boot of linux in Thales-CI.

jquevremont, 2022-09-16:
Done for 64b Linux.
Complex to do for both 32b and 64b Linux as the Linux distro is on the SDCard.

jquevremont, 2022-10-28:
Triggered by the CI, but only for one configuration.
Sébastien: able to load Linux through JTAG. Could it help support several configurations?

MikeOpenHWGroup, 2023-02-07:
@jquevremont, @sjthales, @Gchauvon, what is the status of this issue?

Gchauvon, 2023-02-08:
Done for 64b Linux.
Not started for 32b Linux, it is part of step 1 verification schedule.

jquevremont, 2023-03-14:
The task is tracked in the verification GANTT. Closing it here.

closed 2023-03-14 by jquevremont

JeanRochCoulon, 2023-11-30:
@jquevremont  Can I ask you to review the task done by @Gchauvon and change the status from "Done" to "Finished - approved by reviewer" if you think the task is completed ?

jquevremont, 2023-11-30:
Review performed offline. LGTM