openhwfoundation/cva6 #927: Possible deadlock when executing from NI memory

niwis, 2022-06-30:
This bug was initially reported in #708.

> Consider the following scenario while executing from nonidempotent memory:
> 
> 1. A branch instruction is fetched => `speculative_d = 1'b1`
> 2. The instruction queue is full, `replay` is asserted
> 3. Once the instruction queue is `ready` again, the frontend sends a request to the icache to refetch the branch instruction
> 4. The icache blocks as `dreq.spec == 1'b1` (and we do not fetch speculatively from nonidempotent memory)
> 5. Deadlock

I think there are several possible ways to handle this, for instance:
- Clear the `speculative` register on a replay (proposed in #708)
- Disallow execution from NI memory (is this RISC-V compliant? Are there cases for executing from NI memory, e.g. the debug module?)
_Note: for the debug module, it might suffice to define only writes as non-idempotent and reads (including instruction fetches) as idempotent, which is possible according to the priv. spec but currently not supported by ariane._
- Allow speculative instruction fetches from NI memory, possibly violating the NI requirement
- Preventing replays in the first place when executing from NI memory, i.e. only fetching when the instruction queue can certainly accept an incoming instruction. For NI memory, we generally do not care about performance, and this might be the cleanest solution allowing to execute from NI memory without side effects.

I'm happy to help implement one of the solutions above (or yet another)!

MikeOpenHWGroup, 2023-02-07:
Hi @niwis.  I _think_ this issue was resolved.  Can you confirm?  If so, please close this issue.

niwis, 2023-03-13:
I don't think so. AFAIK one can still configure a memory region as executable *and* non-idempotent, and the core might still dead-lock if one does so.

jzthales, 2024-07-09:
Hello, I am currently working with @cathales in @JeanRochCoulon's team on the superscalar version of the CVA6 at Thales DIS.

On b6a3aa1b037f38313a972b421509bbfe73687beb with these modifications: 

```diff
diff --git a/core/include/cv32a65x_config_pkg.sv b/core/include/cv32a65x_config_pkg.sv
index 10b7e88af..dbab0a00c 100644
--- a/core/include/cv32a65x_config_pkg.sv
+++ b/core/include/cv32a65x_config_pkg.sv
@@ -68,8 +68,8 @@ package cva6_config_pkg;
       PMPEntryReadOnly: 64'd0,
       NOCType: config_pkg::NOC_TYPE_AXI4_ATOP,
       NrNonIdempotentRules: unsigned'(2),
-      NonIdempotentAddrBase: 1024'({64'b0, 64'b0}),
-      NonIdempotentLength: 1024'({64'b0, 64'b0}),
+      NonIdempotentAddrBase: 1024'({64'h0000_0000, 64'b0}),
+      NonIdempotentLength: 1024'({64'hc000_0000, 64'b0}),
       NrExecuteRegionRules: unsigned'(3),
       ExecuteRegionAddrBase: 1024'({64'h8000_0000, 64'h1_0000, 64'h0}),
       ExecuteRegionLength: 1024'({64'h40000000, 64'h10000, 64'h1000}),
```

The coremark test times out:

```
Starting FST waveform dump into file 'verilator.fst'...
/home/jzheng/rhel8/cva6/verif/sim/out_2024-07-09/directed_c_tests/core_main.o *** FAILED *** (tohost = 2147483647) after 2000013 cycles
UVM_INFO  @                    0 ns : ariane_testharness Set UVM_VERBOSITY to UVM_NONE
CPU time used: 411686.14 ms
Wall clock time passed: 409725.46 ms
make: *** [Makefile:407: veri-testharness] Error 255
```

The signal 'speculative_d' (i_ariane > i_cva6 > i_frontend > 'speculative_d') indeed goes high and never gets back to low: CVA6 hangs.

Is it relevant to consider non-idempotence in instruction fetch?