openhwfoundation/cva6 #934: [BUG] Verilator 4.014 error | followed as per travic-ci

jyrj, 2022-07-22:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

I have been trying to compile the simulation files of CVA6 using the ```make verilate``` command. 
I have set everything using the scripts found in travis-ci-emul.sh
But execution of make command throws error as:
```
%Error: /home2/jayaraj/Desktop/cva-6-issue/cva6/corev_apu/src/tech_cells_generic/src/rtl/tc_sram.sv:93: Unsupported or unknown PLI call: $urandom
%Error: /home2/jayaraj/Desktop/cva-6-issue/cva6/corev_apu/src/tech_cells_generic/src/rtl/tc_clk.sv:87: syntax error, unexpected TIME NUMBER, expecting TYPE-IDENTIFIER
%Error: /home2/jayaraj/Desktop/cva-6-issue/cva6/corev_apu/src/tech_cells_generic/src/rtl/tc_clk.sv:90: syntax error, unexpected output, expecting IDENTIFIER or do or final
```
I think it is a verilator version issue. But I have used the same version mentioned in the script. 

  Operating System: Ubuntu 20.04.4 LTS
            Kernel: Linux 5.11.0-27-generic
      Architecture: x86-64

MikeOpenHWGroup, 2022-07-22:
Hi @jyrj, thanks for reporting your issue.  You will need at least V4.100 to compile and simulate the CVA6 with Verilator.

The [README](https://github.com/openhwgroup/cva6#install-verilator-simulation-flow) provides some information about other versions of Verilator that do not properly support the CVA6.

jyrj, 2022-07-22:
The ci/install-verilator.sh script uses: 
```wget https://www.veripool.org/ftp/verilator-4.014.tgz```
Kindly update it, would be helpful in future. 

Also, please suggest a version that would work for verilating. 

[Sorry for label, used default issue template]

MikeOpenHWGroup, 2022-07-22:
I am using Verilator v4.100 and I do not see your issue.  Please give that version a try.

Can you create a second issue for the `install-verilator.sh` script issue?  We try not to mix multiple problems into a single GitHub issue.  Thanks!

jyrj, 2022-07-23:
@MikeOpenHWGroup , the issue is when we use the sources to build (4.110). When I used the .tgz file, changing the version numbers in install-verilator.sh is working correctly and is sufficient to correctly use the verilator. I have successfully compiled the simulation files using 4.110.

jyrj, 2022-07-29:
CVA6 managed to run when I changed the version number found in install-verilator.sh file to 4.110 [and corresponding paths]. Kindly note when I build the same version of verilator from source, it didn't worked. However thank you for porting the cva6 to new verilator.

Now it works perfectly with 4.110.

closed 2023-02-07 by MikeOpenHWGroup