openhwfoundation/cva6 #944: DEAD CODE in CVA6

AyoubJalali, 2022-08-23:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

Hello, @JeanRochCoulon @ASintzoff
While improving the CC of the CVA6 core processor, I notice there is a lot of lines related to the vector extension (V extension) in the id_stage.decoder module, as i know the CVA6 don’t support V extension, so there is any meaning of code that CVA6 can't execute at all ? or we can say it's a DEAD CODE.
[image]

MikeOpenHWGroup, 2023-02-07:
Hi @JeanRochCoulon and @ASintzoff.  I know we have discussed this issue many times.  Have we created a consensus?  I see several options:
1. Mark this "wont-fix" and close it.
2. Add Synopsys pragmas to the RTL to exclude code coverage.
3. Apply the Synopsys code-coverage review tools to identify dead code.
4. Update the RTL to remove the dead code (if it is really dead).

JeanRochCoulon, 2023-03-14:
Yes, this question comes back a lot of time. To have cva6 eligible to certifications, code must be de-activated when it is dead. To answer the @MikeOpenHWGroup 's suggestions
1. This solution is not releant because we would like to achieve 100% code coverage
2. That's the solution I would suggest to adopt: condition the code with XFVEC and add code coverage pragma. Of course pragmas implemented in RTL will be dedicated to verified V5. We pan to modify the RTL to verify another configuration.
3. We have double ckeched with our CAD support, no way to use simulator tool to identify dead code.
4. We prefer to keep all the configuration available in the code.

@ASintzoff are you ok with the strategy ?

ASintzoff, 2023-10-20:
The corresponding code is now filtered by `CVA6Cfg.FpPresent` and `CVA6Cfg.XFVec`.

closed 2023-10-20 by ASintzoff