openhwfoundation/cva6 #960: Unable to view waveform of variable in Questasim

sumandeb003, 2022-09-21:
In corev_apu/fpga/src, the axi2apb  directory contains the axi2apb_64_32.sv. axi2apb_64_32.sv has two variables CS, NS of type enum logic. I am trying to simulate a testbench for this module using Questasim. As shown in the attached screenshot, these two variables CS, NS don't appear in the simulation window. I wish to view the waveforms for these two variables. Can you please advise how to view them in Questasim?
[image]

sumandeb003, 2022-10-06:
Looking forward to your assistance, please...

MikeOpenHWGroup, 2023-02-07:
Hi @sumandeb003, I am closing this issue now.  We do not offer support for commerical EDA tools.

closed 2023-02-07 by MikeOpenHWGroup