openhwfoundation/cva6 #962: CVA6 RTL Templating

JeanRochCoulon, 2022-09-22:
CVA6 is currently configured by:
- ifdef commands to declare configurable Input/Output or structure types
- multi files implementing the same packages or modules. To select CVA6 configuration (CV32A60X_config_pkg.sv,...) or select flipflop or memory for instance.

This methodology can generate bugs. Some tools (formal verification, certitude, code coverage,...) consider dead code as active code,...

In a first step, templating, could be JINJA based, need to be evaluated to replace ifdef and multi packages

yanicasa, 2022-09-28:
We will discuss about it during next core TK on Monday October 3rd.

zarubaf, 2022-09-28:
lowRISC and Opentitan uses [`mako`](https://www.makotemplates.org/) which is very nice. We've also used it quite extensively in the past and it is quite straight forward to use. Very happy to see this happening.

jquevremont, 2022-10-28:
No consensus after Cores TG meeting. Templating on hold. We first should try to get as far as possible with parameters.

jquevremont, 2023-01-20:
Vendorization is a pre-requisite before cleaning up the code.
The team is thinking about a solution to replace the ` directives with parameters; it's not yet implemented.

jquevremont, 2023-03-14:
For step 1, pragmas will be used to disable not-activated lines (e.g. RV64 specifics when Xlen=32) to assess code coverage.
Currently, we have not identified a perfect solution.
Templating is not in the agenda => closing.

closed 2023-03-14 by jquevremont