openhwfoundation/cva6 #969: Boot from sd card failed &  cannot run linux system

StilluSmile, 2022-10-06:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

According to the original FPGA project based on Genesys 2 board, I have implemented CVA6 on my own FPGA development board (also based on kintex7-7k325tffg900-2) because there is no gesysII development board. There is no network port on my own development board, so I removed the Ethernet module in the rebuilt project. 
Execute the following command to write the image to the sd card

sgdisk --clear -g --new=1:2048:6909 --new=2:512M:0 --typecode=1:3000 --typecode=2:8300 /dev/sdb
dd if=cva6-sdk/install64/fw_payload.bin of=/dev/sdb1 status=progress oflag=sync bs=1M
dd if=cva6-sdk/install64/uImage         of=/dev/sdb2 status=progress oflag=sync bs=1M

After Boot from Configuration Memory Device, the serial port prints as follows:
Platform Name             : ARIANE RISC-V
Platform Features         : medeleg
Platform HART Count       : 1
Platform IPI Device       : aclint-mswi
Platform Timer Device     : aclint-mtimer @ 1000000Hz
Platform Console Device   : uart8250
Platform HSM Device       : ---
Platform Reboot Device    : ---
Platform Shutdown Device  : ---
Firmware Base             : 0x80000000
Firmware Size             : 220 KB
Runtime SBI Version       : 0.3

Domain0 Name              : root
Domain0 Boot HART         : 0
Domain0 HARTs             : 0*
Domain0 Region00          : 0x0000000002008000-0x000000000200bfff (I)
Domain0 Region01          : 0x0000000002000000-0x0000000002007fff (I)
Domain0 Region02          : 0x0000000080000000-0x000000008003ffff ()
Domain0 Region03          : 0x0000000000000000-0xffffffffffffffff (R,W,X)
Domain0 Next Address      : 0x0000000080200000
Domain0 Next Arg1         : 0x0000000082200000
Domain0 Next Mode         : S-mode
Domain0 SysReset          : yes

Boot HART ID              : 0
Boot HART Domain          : root
Boot HART ISA             : rv64imafdcsu
Boot HART Features        : scounteren,mcounteren
Boot HART PMP Count       : 16
Boot HART PMP Granularity : 8
Boot HART PMP Address Bits: 54
Boot HART MHPM Count      : 0
Boot HART MIDELEG         : 0x0000000000000222
Boot HART MEDELEG         : 0x000000000000b109

U-Boot 2021.07-rc4-g920075ecfa (Oct 05 2022 - 12:13:58 +0800)

CPU:   rv64imafdc
DRAM:  1 GiB
MMC:   xps-spi@20000000:mmc@0: 0
Loading Environment from nowhere... OK
In:    uart@10000000
Out:   uart@10000000
Err:   uart@10000000
Net:   No ethernet found.
Hit any key to stop autoboot:  5  4  3  2  1  0 
Device: xps-spi@20000000:mmc@0
Manufacturer ID: 9f
OEM: 5449
Name: SD32G 
Bus Speed: 12500000
Mode: MMC legacy
Rd Block Len: 512
SD version 2.0
High Capacity: Yes
Capacity: 29.1 GiB
Bus Width: 1-bit

MMC read: dev # 0, block # 1048576, count 20480 ... 20480 blocks read: OK
## Booting kernel from Legacy Image at 90000000 ...
   Image Name:   CV64A6Linux
   Image Type:   RISC-V Linux Kernel Image (gzip compressed)
   Data Size:    10004435 Bytes = 9.5 MiB
   Load Address: 80200000
   Entry Point:  80200000
   Verifying Checksum ... OK
## Flattened Device Tree blob at bffe08d0
   Booting using the fdt blob at 0xbffe08d0
   Uncompressing Kernel Image
   Using Device Tree in place at 00000000bffe08d0, end 00000000bffe551f

Starting kernel ...
........
[   18.752860] libphy: Fixed MDIO Bus: probed
[   18.752860] libphy: Fixed MDIO Bus: probed

[   18.783619] lowrisc-digilent-ethernet: Lowrisc ethernet platform (30000000-30007FFF) mapped to ffffffd004028000
[   18.783619] lowrisc-digilent-ethernet: Lowrisc ethernet platform (30000000-30007FFF) mapped to ffffffd004028000

However, the print output of the serial port has been stopped at "lowrisc-digilent-ethernet: Lowrisc ethernet platform (30000000-30007FFF) mapped to ffffffd004028000" for almost more than 3 hours, and it seems that the normal startup has not been completed.
Has anyone else encountered this problem? Thank you very much for your answers and replies~

maybe I missed some setting or modification. I wonder where the problem is. Both fw_payload.bin and uImage used are generated in cva6-sdk.

eyssartk, 2022-10-06:
Hello, 
Thank you for your interest in the CVA6.
It seems you almost reach the end of the linux boot process.
If I understand correctly, you do not have the low-risc ethernet module. So I think you should try to remove the ethernet module in the dts, located in u-boot/arch/ricv/dts/cv[32|64]a6_genesysII.dts.
Good luck !

StilluSmile, 2022-10-07:
> Hello, Thank you for your interest in the CVA6. It seems you almost reach the end of the linux boot process. If I understand correctly, you do not have the low-risc ethernet module. So I think you should try to remove the ethernet module in the dts, located in u-boot/arch/riscv/dts/cv[32|64]a6_genesysII.dts. Good luck !

thank you very much for your suggestion. Issue solved for me

StilluSmile, 2022-10-07:
> The file system size is computed to fit all the things you put into the image. You can put your program or benchmark in the cva6-sdk/rootfs folder and you will find it in the image. The same is done for cachetest and tetris programs.

thank you~ I will try it

closed 2022-10-07 by eyssartk