openhwfoundation/cva6 #971: Does the feature/gsoc branch still have a purpose?

MikeOpenHWGroup, 2022-10-07:
### Is there an existing CVA6 bug for this?

- [X] I have searched the existing bug issues

### Bug Description

The CVA6 branch `feature/gsoc` has gone stale and is also being flagged for a security alert.  The offending file is [docs/Gemfile.lock](https://github.com/openhwgroup/cva6/blob/c9537f822113ba51a740201fb5ad0f2fa946c1ea/docs/Gemfile.lock#L1) (see below).

As this branch has not been updated in 3 years and it has the security alert, I propose we delete it.  @zachzzc, @zarubaf, @msfschaffner is there any reason why this branch _cannot_ be deleted?

[image]

MikeOpenHWGroup, 2023-02-07:
@zachzzc and @zarubaf, this issue has had no response in four months and the `feature/gsoc` has not seen any activity in that time.  Please speak now, or I will delete that branch in one week (by 2023-02-14).

zachzzc, 2023-02-11:
Hi Mike. Feel free to remove the branch. I don't think it is currently used by any one.

zarubaf, 2023-02-11:
*Please do not delete this branch!*

Jbalkind, 2023-02-18:
Is this the branch that has the code for dual issue CVA6? I'd prefer we didn't lose it if so

MikeOpenHWGroup, 2023-02-27:
@zarubaf, @Jbalkind, can you provide a some insight as to the purpose of this branch.  I am happy to keep it if needed, but if we are going to keep three-year-old, stale branches, let's record the reason.  It would be good to place it at the top of the README.

Jbalkind, 2023-02-27:
If it's the branch I'm thinking of then you can see the info here: https://zachzzc.github.io/gsoc-ariane/

zarubaf, 2023-10-20:
The branch is still valid, I don't see a good reason to delete it.

closed 2023-10-20 by zarubaf